Ralim/IronOS #675: Flashing problem

ligi, 2020-07-28:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**

getting this when flashing:
```

✗ ~/g/3/t/Flashing (master=) 
./flash_ts100_linux.sh /home/ligi/Downloads/TS80_EN.hex 

#####################################################
#   Waiting for TS100 config disk device to appear  #
#                                                   #
# Connect the soldering iron with a USB cable while #
# holding the button closest to the tip pressed     #
#####################################################

Found TS100 config disk device on /dev/sda
[sudo] password for ligi: 
Mounted config disk drive, flashing...
'/home/ligi/Downloads/TS80_EN.hex' -> '/tmp/ts100/ts100.hex'
Waiting for TS100 to flash
Remounting config disk drive
Flash result: 
/tmp/ts100/ts100.err
✓ ~/g/3/t/Flashing (master=) cat /tmp/ts100/ts100.err                   00:32:02
cat: /tmp/ts100/ts100.err: No such file or directory
```

ligi, 2020-07-28:
fun fact - I changed the script afterwards to:

```
✓ ~/g/3/t/Flashing (master *=) git diff                                                                                00:40:40
diff --git a/Flashing/flash_ts100_linux.sh b/Flashing/flash_ts100_linux.sh
index 2179ee8..9215dd1 100755
--- a/Flashing/flash_ts100_linux.sh
+++ b/Flashing/flash_ts100_linux.sh
@@ -127,4 +127,4 @@ wait_for_ts100
 mount_ts100
 
 echo "Flash result: "
-ls "$DIR_TMP"/ts100*
+cat "$DIR_TMP"/ts100*
```

to try to figure out what the error was (guessing the temp file was deleted) - but interestingly then flashing worked ..

Ralim, 2020-09-14:
@ligi 
Going to be honest and say I haven't seen this before; though the DFU bootloader is rather flaky and even rebooting a computer will make a difference :/

pliski, 2021-05-02:
> fun fact - I changed the script afterwards to:
>
> ```
> ✓ ~/g/3/t/Flashing (master *=) git diff                                                                                00:40:40
> diff --git [...]

Worked for me too!

In fact the device unmounted itself after the flash attempt ... but somehow after the cat mod it worked ... a sync problem maybe?

closed 2021-06-25 by Ralim