Ralim/IronOS #677: TS 100 not heating up (Display fail)

NotBitten, 2020-08-11:
Hello, i got my TS 100 today. I opend it, installed the soldering tip and wanted to start soldering. But when i plugged it in and pressed the A Key it showed that it is heating up for a moment but after half a second or less it returned to the main screen and didnt heat up. I dont know how to fix this problem and I didnt find anything on the internet.

Please help me!

whitehoose, 2020-08-11:
What are you using for power?
https://oscarliang.com/ts100-soldering-iron/ if you look down the page there is a manual https://cdn.shopify.com/s/files/1/0948/3076/files/TS100_Soldering_Iron_Instruction_Manual.pdf?188285643864339764

showing those important little places - grubscrew 1 fixes the bit in place - but on a new iron it will (possibly) be screwed home and prevent the tip inserting fully home and snug - unscrew it and try re-seating the tip - Interested to see how many ppl re-tighten that fixing screw Vs those who just rely on spring friction to keep the tip in place.

Finally - plug into a power source that's >12v up to 24v (shoot for 18 to 24).

I have a ts80 so I'm maybe just slapping my gums - a LED switched PSU from ebay seems like a cost effective option - you can get a 24v 120w psu for about £12, tons of fan cooled power for not a lot of money - (you'll still only get 65w max for the iron) - but the PSU should run nice and cool without stress, or get a 200w+ and a 5A-10A buck converter and you can have a cheap bench power supply for a couple of quid more.

Add an xt60 connector and a 5 or 6s lipo battery and you are road ready.

discip, 2022-04-23:
@NotBitten
Is this issue still present with the latest release?
Otherwise please close this issue. 😊

thanks

closed 2022-05-08 by Ralim