Ralim/IronOS #678: SQ001 support request

1ofko6, 2020-08-12:
I would like to use this fw with SQ001 too..
Please tell me if it is posible

Ralim, 2020-08-12:
Hello,

The SQ001 is a clone of the TS100 so it should _just work_.
However, as they do not release schematics I cannot _guarantee_ it will work.

lamosca01, 2020-08-15:
It works fine on clone SQ001. You have to calibrate it. Regards!

1ofko6, 2020-08-15:
Nice, but how can i calibrate it? If you mean temperature, i have nothing where can i measure temperature😅
Of course TIP gain i dont know what i change if i change this setting

Some idea: to improve main screen while soldering: can you add here i thin in the top left coner the watts? 
Maybe auto-detection of battery cells would be fine..
Im not using always 6s🤷🏻‍♂️😃
Thank you

whitehoose, 2020-08-15:
@1ofko6 
I'm not having a pop at you - or any of the other's who seem to put so much emphasis on the accuracy of that temperature reading.  I can sort of see, if it says 10 of course it should deliver 10.  That's what techie types live by (to four decimal places :grin:).  But when it comes to soldering, I've never in my 50+ years of burning fingers and melting insulation looked at a job and thought that's going to take 215.6 degrees c.  Nearest I've ever got is "that could probably do with a couple of clicks less (or more)".  When I started soldering - that equated to leaving the tip on the stove for a bit longer.  I've only had a thermostatic tip for about 3 years, before that I used a range of sizes and weights of fixed wattage irons and my eyes, the temperature was never a numeric value, it was chosen on the strength of how much copper was involved with the area I was going to melt solder onto.  When melting solder - I'm looking for that ripple you see when the surface melts - or the shape of the cone the solder takes when it melds with the component lead, There is rarely any mistaking if it has or hasn't taken but I've never studied the numbers.  I can tell you I usually shoot for 320c, just how close the irons with numbers actually are I have not a clue - but I know when I touch the test piece what I need to do to make my working setting goldilocks.

If there was a scale of 1-10 all I'm looking for is how many more or less does it need?  If you go too far will the track lift or will the part turn into a sticky mess.  Once the tip is set, even outside, more often than not I don''t need to alter the temp again.
The Iron is really only designed to maintain a little consistency ....  a given thermal mass of tip can only maintain the temp if you don't dip it in ice or try and heat a copper brick - so you are going to see some swing in operating temperature - all you can hope is that it returns to "normal" consistently.

1ofko6, 2020-08-15:
I dont need high accuracy temperature🤣
I know it is just piece of information😃😉
But the baterry cell count detection on the start would be good, wouldnt it?
I mean.. if i set 4S it will store in eeprom, than, if i connect 6S it is still working with 4S so my 6S battery will be drained and possibly destroyed🤷🏻‍♂️ so i must care about that, if it is 4S or 6S.. autonomous selection would be great😃
What do you think?
Have a nice day😉😃

Ralim, 2020-08-18:
@1ofko6 

> to improve main screen while soldering: can you add here i thin in the top left coner the watts?
If you use detailed mode it will show the estimated power

>  Maybe auto-detection of battery cells would be fine..

This is not overly possible, as voltage ranges for cells overlap

Lithium Polymer cell range = 2.5V-4.35V

| Cell count | V_Min | V_Max |
|------------|-------|-------|
| 3          | 7.5   | 13.05 |
| 4          | 10    | 17.40 |
| 5          | 12.5  | 21.75 |
| 6          | 15    | 26.1  |

The min and max levels overlap, so its not possible to tell between cell counts correctly.
I.e. if the voltage reads as 19V, is it 5 or 6 cell battery?

1ofko6, 2020-08-18:
I know about detailed screen, well thats okay😉
About liion batteries, well you are right, but for practical purposes you need to use narrower range of voltage for one cell.
I mean most people use Lipol, liion are not so powerfull and lihv are not easy to get in pack for drone, lihv are not that powerfull..
So in reality you have range of 3.5 - 4.2
Well to cover some miss calculation im using 3.5 - 4.30
This is much better, isnt it?
In reality most people connects fully charged battery, so as a reference voltage for diving the main voltage to get cell count, 3.9V per cell is okay i think, maybe 3.85 could be better but i dont know, its rather for real testing..
Thats how i would program that..
I can write that program but i cant compile that and than test it soo...🤷🏻‍♂️
Have a nice day and thayn you for your time&work  on your firmware😃

Ralim, 2020-08-18:
> About liion batteries, well you are right, but for practical purposes you need to use narrower range of voltage for one cell.
Ah but remember about voltage drop in cables + sag on smaller batteries as well
So I use a voltage of 2.5 to allow for an approx ~3.1V unloaded. :smile: 

Indeed it is possible to "guess" based on 3.9V, but there are users who are using already "nearly flat" drone batteries, mixed with people using fully charged ones, so we can expect either 4.15+V or < 3.2V which leaves us in the same problem of having potentially confusing states, so its far simpler for people to set a known cell count :D

1ofko6, 2020-08-18:
Well now days small batteries are away i think
So liion is better to discharge to 3V on 2.8 you are destroing it..
Of course in program there could be both, manual and auto cell detecrion while program running just use higher priority on manualy set cell count
And in deeper setings could be the reference voltage settings for individual needs
Mean: 3.9 for me, because i use charged baterries
 And 3.5V for others🤷🏻‍♂️
Well this seting should be with steps on 0.05V
Maybe from 3V to 4V
I know that there could be some miss calculation, but it could be really nice feature😉
Like.. in betaflight

Ralim, 2020-09-14:
Betaflight makes the same assumption you to -- that people _start_ near full charge. Which is not correct for all users in this case.
I only have found this out, thanks to people raising an issue when it did try to auto guess because they were using packs at ~30% because they had already flown their craft.

Now there is probably a middle place where there can be an "auto" position in the cell selector menu... Which should be quite doable :)

Kealper, 2020-11-06:
SQ-001 owner chiming in: This flashed and worked perfectly without any issues on my Sequre SQ-001, just follow the instructions for the TS100 and it'll work just the same. I did have to calibrate the input voltage though, mine was reading about 0.7v too low initially.

closed 2022-02-15 by Ralim

KaMyKaSii, 2022-06-30:
And about the Sequre si012, has anyone tested it?

Danook2021, 2022-07-25:
I have broblume the tip not working if i conect it the sq-001 say No tip... No way to change satting it is, stop... I tay other tip it is work...