Ralim/IronOS #681: My ts100 broke

Borcemitev, 2020-08-24:
Hello 
I accidentally broke my ts100.. I was powering it via lab bench power supply and set it to 30v and it broke.. I opened it up and on a first there wasn't anything looking faulty but testing some component while on power there was smoke.. I don't know now which components i should change... Here are pictures from the faulty one i can find.. 
[image]
[image]

whitehoose, 2020-08-24:
It's a simple answer to a simple question (although the actual process may not be quite so simple).

"I don't know now which components i should change..." - that's easy - you change the faulty ones. :grin:
The devil of course in the detail of just how you determine which components are faulty.  There isn't a page on the internet that can tell you which components will be faulty - it may be one resistor - it may be every IC on the board (are you feeling lucky?) you have to find them yourself.  to help download a circuit diagram here: https://github.com/Ralim/ts100/issues/449

Then you have two approaches - hem, Ok ... 3 ... well 4 then.  
1. Easiest is buy a replacement.
2. Contact mini support and as to buy a replacement mainboard (my experience is they could well give you one free!).
3. Make a list of every discoloured component and go to your favourite component site and buy replacements - (hint) you can buy SMD resistors in "books" of 2000 for very little cash.
4. The best way IMHO is break out the multimeter and (referring to the diagram) and (starting at the power connector) test your way round the board,  Components in place can be misleading test onboard, then if its not what you expect, remove the component and test again.
REMEMBER - a multi-meter uses a battery - when testing ICs etc bear this in mind - don't do more damage when testing.
It's a good way to learn ... you have to be logical, don't go random, follow the tracks.
The biggest drawback is once you identify a fault ordering parts that may take 90 days to arrive.

Over to you

phedders, 2020-08-24:
On Mon, 2020-08-24 at 03:29 -0700, whitehoose wrote:
> It's a simple answer to a simple question (although the actual
> process may not be quite so simple).
> "I don't know now which components i should change..." - that's easy
> - you change the faulty ones. 😁
> 
> The devil of course in the detail of just how you determine which
> components are faulty.  There isn't a page on the internet that can
> tell you which components will be faulty - it may be one resistor -
> it may be every IC on the board (are you feeling lucky?) you have to
> find them yourself.  to help download a circuit diagram here: #449
> Then you have two approaches - hem, Ok ... 3 ... well 4 then.
> 
> Easiest is buy a replacement.
> Contact mini support and as to buy a replacement mainboard (my
> experience is they could well give you one free!).
> Make a list of every discoloured component and go to your favourite
> component site and buy replacements - (hint) you can buy SMD
> resistors in "books" of 2000 for very little cash.
> The best way IMHO is break out the multimeter and (referring to the
> diagram) and (starting at the power connector) test your way round
> the board,  Components in place can be misleading test onboard, then
> if its not what you expect, remove the component and test again.
> 
> REMEMBER - a multi-meter uses a battery - when testing ICs etc bear
> this in mind - don't do more damage when testing.
> 
> It's a good way to learn ... you have to be logical, don't go random,
> follow the tracks.
> 
> The biggest drawback is once you identify a fault ordering parts that
> may take 90 days to arrive.
> 

Option 5 = (1|2)+(3|4)

Get a second iron or MB, then us it to compare the values of all
(passive) components in circuit to find our where the pesky electrons
go missing or hit a dead end. Once repaired you then have knowledge,
and a spare to keep in another kit.

whitehoose, 2020-08-24:
Just seen ph's reply (lol)

Looked at the pikky with the ringed component - to me that looks like a reasonably OK rectum-fryer.  They tend to be dull badly lettered grey blobs, but also they tend to be quite high current little buggers.  Look and test round the power jack, the pixies tend to run in straight lines so the first components they touch tend to take the worst of the damage.

It gets "interesting" round ICs and if you look you'll see a variety of sizes of components too to complicate things, 30v is quite a slug though, you should expect a little smoke.

4nthonylin, 2020-08-26:
Could you elaborate on broken? Won't power on and power supply current limits the device?
Taking a look at the schematics. Here are some components for concern when exposed to 30V:
## Components
* High: D3 - BZT52B20S
D3 is a zener diode that clamps to 20V.
[image]
Based on the path, Vin = 30V, Vfd4 = 1V, Vin to the switching regulator will then be 20V we can calculate current through R30. 29 - 20 = 9 Volts over the resistor. V/R = I so 9/150 = 0.06A Since current is the same through a branch, D3 will be seeing 0.06A as well. Calculating the power it has to dissipate = 20V * 0.06A = 1.2Watts. According to the [datasheet](https://static6.arrow.com/aropdfconversion/73bdfe1957e99f28453af487636b05b6a577ae98/bzt52b2v4s20series_g1804.pdf). The absolute max power this diode can handle is 0.2 watts. We are now 6x over the absolute max power. This is most likely the diode that you circled causing the smoke. This diode should now fail as a short to ground now which is most likely why your soldering iron doesn't work anymore.
* High: R30
Same as above, the chip resistor is most likely a 1/8 watt chip resistor. It experiences 9v * 0.06A= 0.54 Watts of power. Most likely that resistor is burnt out too now. Furthermore, once D3 has shorted, this resistor will now see a 29V drop which means 0.2A of current or 5.5Watts of power now most likely smoking it now. 
* Medium: D4
Same as above, once D3 has shorted, we will now see 0.2A with 1volt drop. According to the [datasheet](https://www.onsemi.com/pub/Collateral/1N914BWT-D.pdf), this diode should be able to withstand 0.3A however I'm nots ure if

## Potential Fixes
1) D3 is only really a protection diode and not critical to the operation of the device. I would first remove this device from the pcb. Most likely the circled diode but you should confirm to see if the anode is grounded (the side without the line marking on it).
2) I would check the resistance of R30 to check if it's still 150 Ohms. The resistor should the labeled with an 18A.
3) If resistance still checks out, you should be able to try to apply 15V to the device and see if it powers up. I checked the switching regulator and it should handle up to 36V input

whitehoose, 2020-08-27:
Rather than calculating the likely fail points - it's much easier (and better practice) to dust off the old AVO and poke some copper, that way you know for certain.  I don't dispute your maths (or logic) @4nthonylin but them pixies are unpredictable when stressed.  My guess is @Borcemitev could the the practice.

With a 28day lead on component delivery times (if you don't have a well stocked box of bits) it's going to be a long fix if something else followed the diode fairy.

@Borcemitev, thinking about it it's a good idea to open a dialogue with Mini while you follow any other routes that appeal, that way you have a plan b.  Mine is a TS80 - it failed within a week, I spoke to bangood who replaced it, said to keep the corpse.  Sad to see the empty shell laying there I contacted MINI to ask about spares (I was honest with the description) mini wanted to see the damage and offered a replacement mainboard.  I posted the old one, they posted the new mobo (airmail), within 3 weeks from the contact I had 2 working irons!  Their customer service is very good and from my experience they are more than generous with parts and help.
Worst they can do is say No (I doubt they will) and if you dont have a well stocked shelf it beats heck out of buying 1 spare at a time.

closed 2020-12-13 by Ralim