Ralim/IronOS #682: Add standby mode to iron

GreekGeorge, 2020-08-30:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [X] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Feature, please

* **What is the current behavior?**
No way to set an idle temp afaik

* **What is the expected behavior?**
By pressing a button, to force idle/sleep mode, and after a given amount of time, let the accelerometer detect motion to bring it back out of idle/sleep mode. Sometimes you just need to set down the iron for a couple of minutes to place some components, and then go back to soldering. If you could set the iron into a sleep mode, and drop the temp to >200°C, you will expand tip longevity as well.
Concept:
1. Press 'B' button, and set down the iron
2. Accelerometer is bypassed for 5-10sec (time it takes to set down the iron onto a stand)
3. Temp drops to ~175°C (standby mode)
4. If the accelerometer notices movement, it ramps back up to soldering temperature
5. If the accelerometer doesn't notice movement for a set amount of time, the iron goes into sleep mode (as done currently).

Ralim, 2020-08-30:
Hello,
I'm not sure what this is asking for really that wouldn't be achieved by the normal default behaviour of the firmware?

Just set the sleep time to like 30 seconds and this should all basically work out of the box?

GreekGeorge, 2020-08-30:
Setting sleep mode as low as possible isn't exactly the same as what I'm describing imo, as sleep mode would activate after remaining motionless, where as an idle mode as I'm describing it will be set by the user on demand. This would be closer to what more well-known companies like jbc and Weller do when the iron is placed in the holder.
As mentioned before, idling at a lower temperature is very important for expanding the tip's longevity.

whitehoose, 2020-08-30:
If it's what I have on my bench - it works from an embedded magnet in the handle (easy to do with a (relatively) big unit like a weller) It's quite big because the neo magnet closes a reed fixed inside the stand.  I've seen a smaller electronic one on a friend's (far) more costly setp - but it's still quite a size compared to a ts 100 or 80 (and you'd need a dedicated stand or similat - so may as well buy a weller (IMHOI its a better iron too :grin:).

The sleep "just works" by leaving it alone for x seconds, no extras required - whereas an "I've just put it down" detector needs a dooh dah to interact with.
You could alternatively cobble something together with a PIR or radar detector or some sort of IR detector focused on a particular area of your bench but when the iron went cold??  I can think of a few ways, but they would all be "projects" and not at all slick

GreekGeorge, 2020-08-31:
[image]
JBC shorts a pin through the tip (the tip is grounded), and ramps down to - as they call it - sleep mode.
This is obviously not possible with the TS100, as it's a AIO portable soldering iron. As whitehoose mentioned above, there are other alternate ways to achieve 'sleep mode', none of which can work with the TS100 without modifying a reed switch, or else IR detectors or PIR. That's why imo the "press B to put it in sleep mode manually, and then when it's picked up again it'll heat back up" is the best solution to the problem, and beats setting sleep mode very low etc.

GreekGeorge, 2020-08-31:
TS100 listings are slowly being removed, for both tips and the actual iron. Expanding tip longevity should & could be one of Ralim's key factors over the stock firmware, appart from all the additional features.

Ralim, 2020-08-31:
@GreekGeorge 

Improving the life of the tip _is_ one of the first things I focused on, which is why the sleep modes are easily adjustable. You don't need to best a dead horse here.

I don't get why this is overly desired, especially as it would require removing another existing button shortcut.

If you want instant sleep when you put it in the holder, using the IMU is the best solution with the existing hardware as it requires no extra steps, and it can kick in, in only a few seconds.

The Pinecil is coming from pine64 which may have more features in this department, but im not certain on anything in that regard so far.

Ts100 is not going away anywhere, though its peak sales volumes have probably occured (saturating market) so probably slightly less carriers out there, but I wouldn't be worrying, there are _lots_ of tips out there already.

GreekGeorge, 2020-08-31:
I mean no offense to the work you've already put in, just giving my own $0.02. I'll use sleep mode more frequently.
Thanks for your time.

closed 2020-08-31 by GreekGeorge