Ralim/IronOS #688: Voltage-dependant power limits?

EternityForest, 2020-09-05:
* **I'm submitting a ...**
  - [ ] Bug report
  - [ X] Feature request
  - [ ] Translation

* **What is the current behavior?**

Right now, power limiting is a fixed setting that you control manually

* **What is the expected behavior?**

Instead of an on/off control, provide a voltage below which the limitations are enabled

* **What is the motivation / use case for changing the behavior?**

USB-PD triggers only go up to 20 volts, providing an easy way to distinguish them from the dedicated 24v chargers, presumably bought for the purpose and chosen to provide the full 65 watts.

There are even USB 5v to 12v boost converters, which could likely provide a bit of power in an emergency.

In addition, it appears from the specs that the Bosch heated jacket controller can actually power this, but it's rated current is only 1.2A  As 14W is still a usable amount of power, supporting this well would make the TS100 one of the most convenient rechargable irons out there!

Thanks for the awesome project!

Ralim, 2020-09-06:
Hello 🙂 

So two thoughts on this:

## A - UI

How would you think to expose this on the menu system that is not confusing or hard to configure?

## B - Technicality

Because of the way these irons work, the power limit is an _average_ power limit, it does not limit the peak power drawn as the tip itself can only be "on" or "off" and we are duty cycling it for control.

As such, a common misunderstanding is that if you have a power supply that does 20V @1A, we could limit to 20W and it will work out. However this will often cause the power supply to either trip over current protection or heat up due to the "on" still being 20V/4.5ohms = 4.4A spikes, and we are just controlling the average. This is especially true on _some_ PD supplies that are picky about overcurrent limits.

That said, no reason _not_ to add this, so long as people would have a need :)

paulfertser, 2020-09-11:
I'd say having power limit work for input voltages less than 23 V (hardcoded) would cover pretty much all reasonable usecases; I can't imagine how it can happen that someone would be using a 24 V PSU that's not powerful enough for the iron so that it requires the limiting.

Ralim, 2020-09-11:
😅 
I feel like that is just as risky as any other randomly picked threshold though, what if your iron is mis-calibrated... Or your like me and have used a 24V 1A power supply in a pinch because it was nearby (not reccomended) 🤣

EternityForest, 2020-09-11:
There's also a lot of 19v adapters in use too, if I understand the way
others are powering these.

It might be way more complicated than needed and totally not worth it, but
maybe the power limit enable could become on/off/auto, with a lookup table
like:

0-15v: Greater of user selected power or 10W(1 amp supplies are easy to
come by)

15-19v: 15W (Again, 1 amp, although this voltage range is pretty rare)

19-19.5v: The greater of user selected power or 30W(1.75A laptop adapters)

19.5v-23v:  User selected or 45W, assume this is a USB-PD to 2.5mm adapter

23v and up: full power, assume adapter choosen specifically for the purpose.

paulfertser, 2020-09-11:
My suggestion was based on an assumption that it's rather likely a soldering iron user has a laptop PSU handy, and they all give 40+ W power on 19 V which is enough for almost any soldering work. And that those users are ever going to use a 24 V source only to go higher than that.

Ralim, 2020-09-11:
It might be simpler to do the phone style auto-detect and increase ramp the load up until we see a drop on the supply voltage?
This would work on supplies that dont instantly shut-down at least. (Not tested).

discip, 2022-04-23:
@EternityForest
Since this is quite old, I wonder if anyone is still interested in such a feature.
Maybe this should be closed. 😊

thanks

closed 2022-05-08 by Ralim