Ralim/IronOS #693: Power limit is not doing justice to certain PSU

paulfertser, 2020-09-06:
Observations:

Using a small PSU for Toshiba ac100 netbook rated for 19 V 30 W output. Original miniware tips: TS-I (measured 7.4 Ohm at room temperature with UNI-T T60A DMM) and TS-BC2 (7.6 Ohm). PSU output voltage is 19.0 V and the firmware is calibrated to it.

With TS-BC2, tipResistance = 70 and no power limit heating up works normally and the detailed screen shows up to ~47 W. With TS-I heating up leads to reboot unless power limit is set to 20 W or less (with unmodified tipResistance it required as low as 10 W).

Speculation:

Output PWM frequency is too low to usefully limit power drawn from PSUs with small output capacitors. I'd like to be able to play with the duty cycle but calculations involved are not outright obvious to me.

  - Firmware Version: d75f361c4e274192dae5f7f1c16b7b46da296fbe
  - PCB Version: TS100, v2

Ralim, 2020-09-06:
Hello,

So there are two stacked timers for the PWM (fast and slow), where the fast is basically on/off and is used to feed a square wave to the tip (control is ac coupled so that firmware lockup doesn't leave tip on). This timer is fixed at a moderate rate (but not overly fast), as using higher speeds results in worse performance of the control of the tip MOSFET (and is audible on some units thanks to ceramic capa).

The lower speed timer is the actual pwm for the tip control. This is the one that is modulated to control the average power to the tip.

Because the ADC cannot read the tip temperature while the tip is on (due to It using the same connections) the pwm cycle is deliberately slow so that we can insert a fixed off period to allow op-amp to stabilise (come out of saturation) and then the ADC takes 8 samples of the tip temperature and then the PWM resumes.

The ADC completing these samples is what sends the notification to unblock the PID task.

Which means the PID is always running effectively one PWM cycle behind.

Some of this is documented over [here](https://ralimtek.com/stm32_double_pwm/)

So there are a couple of hardware reasons why the pwm is so slow, which is why the power limiter is an _average_ limit, not a peak limit. 😅

paulfertser, 2020-09-11:
Thank you so much for a rather elaborate explanation, now the code is pretty clear! I do not have a 'scope and disassembled iron handy so please excuse me for not measuring the real timings. "output < 230" is arbitrary, based on the idea that 255 in "fast" mode corresponds to 240 in "slow", so it's just 240 minus "some".

Ralim, 2020-09-11:
Glad it helped make sense :) 

I'll leave discussion of the PR over there, but this feels logical to me.

At one point the whole control loop was a lot faster that helped with this, but that broke on the non-miniware irons as they had a much longer settling time on the op-amp. Though that was ages ago :D

Very keen to give it a go and see how it performs :)

Trying to find a nicer means of determining the magic value of 230 might be good, though no ideas come to mind :)

PR #697 (2020-10-11): [PoC] limit maximum power more efficiently
With this a TS-I tip is usable with a small netbook 19 V / 30 W PSU with
power limit set to 40 W (38.9 W is reported during the heating up
stage). Without this the device just reboots on attempt to turn on the
heater (unless the power limit is set to 10 or even 5 W).

This code doesn't affect maximum power available and allows up to 73 W
when a beefy 24 V / 96 W PSU is used.

Should be useful for all models, not just TS100.

The fixed comments are based on calculations, not measurements!

Supposed to fix #693.

Please try and fill out this template wh [... cut]
commit 30be5e0d (2020-10-11): Introduce PWM freq switching so that power could be averaged on shorter intervals

With this a TS-I tip is usable with a small netbook 19 V / 30 W PSU with
power limit set to 40 W (38.9 W is reported during the heating up
stage). Without this the device just reboots on attempt to turn on the
heater (unless the power limit is set to 10 or even 5 W).

This code doesn't affect maximum power available and allows up to 73 W
when a beefy 24 V / 96 W PSU is used.

Should be useful for all models, not just TS100.

The fixed comments are based on calculations, not measurements!

Fixes #693.
closed 2020-10-11 by Ralim