Ralim/IronOS #694: Sub-menus layout and other small bugs

federicodilo, 2020-09-07:
_These are [Bugs]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

1) "Tip gain" setting contain a unique sub-menu layout.
1.1) Is the "Tip gain" value in dB?

2) "Tip model" setting is not shown in the menu.

3) "Power Pulse W" shows a check box whereas is not needed.

* **What is the expected behavior?**

1) The value of "Tip gain" setting could be show directly on the most right part of the screen.
Consider also that a sub-menu layout would be useful also for "Power Pulse W", "Temp change short", "Temp change long" since they require multiple presses in order to be deactivated.

* **Steps to reproduce the bug:***

- Turn on
- Go to Settings>Advanced settings>Tip gain
- Go to Settings>etc.

* **What is the motivation / use case for changing the behavior?**

Improve usability/even out settings

* **What are you running:**

  - Firmware Version: 2.11
  - PCB Version: TS 100 PCB 1

Ralim, 2020-09-07:
Hello

- 1 I'll look at this one again

- 1.1 it's in uV/deg C

- 2 this was removed over 6 months ago

- 3 this is a bug :)

federicodilo, 2020-09-07:
Sorry @Ralim, by flashing the last release I saw it was replaced with "OFF", now it's fine.

discip, 2020-09-08:
@Ralim

**1.** I just closed **[#642 consistency issue](https://github.com/Ralim/ts100/issues/642)**, but somehow I missed the **1st** item @federicodilo pointed out.

**2.** Did this become redundant?

Ralim, 2020-09-09:
I'll look at 1 when I can :)

2 was removed as it was redundant

Ralim, 2020-12-30:
Going to close this out as I think they are all resolved at the moment.
Yell if not and happy to continue on anything :)

closed 2020-12-30 by Ralim