Ralim/IronOS #696: iron heats up in `[S and O]` - mode without being moved

discip, 2020-09-08:
_This is a [Bug Report]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
If **`Auto start`** is set to either ***S*** or ***O***, the iron starts to heat up, without being moved.

* **What is the expected behavior?**
If **`Auto start`** is set to either ***S*** or ***O***, the iron should start heating up only, if it gets moved.

* **Is the issue reproducible.**
Yes

  * ***Steps to reproduce:***
    1. Go to **`Soldering settings`** and set **`Auto start`** to either ***S*** or ***O***.
    2. Unplug the iron.
    3. Fix the iron so it won't move, when being plugged in.
    3. Plug it in again and watch it to start heating up, although it should *"sleep"* until it gets moved.

* **What is the motivation / use case for changing the behavior?**
To help, making the iron function as it was intended to.

* **Additional information**
     ### In the **`info menu`** *(holding B in idle screen)*, I read the output of `Move` and to me it seemed suspicious (as if *time* and *movement* were interlinked), because if the iron is moved the value increases slightly. But when the iron is left for a while and then moved again, the value increases drastically. The longer you wait the bigger is the increase.

* **What are you running:**
  topic|**TS80**|**TS80P**
  -|-|-
  Firmware Version|**v2.12.D75F361**|**v2.12.D75F361**
  PCB Version|**2**|**2**
  Power Supply|**SDC-18W** *(stock)*|**QC09** *(stock)*

Thank you in advance
kind regards

Ralim, 2020-09-11:
@discip 

You are on the right track for the movement timer.
If you sit the unit stationary the movement timer should *not* updated.
its the timestamp as of the last movement, so whenever you move it should update; and when the unit sits still it should stay static.

> If Auto start is set to either S or O, the iron should start heating up only, if it gets moved.

In S mode iron should heat up to sleep temperature, then when the iron is moved it should jump to the final temperature.

Out of curiosity, are you plugging in the unit while holding it and then setting it down - or are you powering up a power supply that would not cause cable movement?

discip, 2020-09-11:
@Ralim
Good afternoon,
Yes, I now the thing about the S mode.
I put the iron stationary and I am very carefull about the cable as well, so there is no movement, that could trigger the accelerometer.

paulfertser, 2020-09-11:
@discip so what Move values are you seeing? I have "20" here if I'm trying to be a bit careful when giving it the power and checking right after it boots. Both S and O work as expected. But it's on TS100, not TS80. Also, what sensitivity do you set it to?

paulfertser, 2020-09-11:
The Move value shown is the _timestamp_ of the last movement detected. So no wonder it's related to the time.

paulfertser, 2020-09-11:
@discip , please enable "detailed soldering screen" and "detailed idle screen" and retry.

discip, 2020-09-11:
@paulfertser
I am seeing different values dependent on the model.

I am using the *detailed screen* for both.
-|TS80P|TS80
-|-|-
values|"13" & "20"|"37", "39" & "46"
sensitivity|"7"|"7"

discip, 2020-09-11:
@paulfertser
No, it does not go to full soldering mode right away. It first shows *"Sleeping ..."* and almost immediately after that starts to heat up.

On the **TS80** there is *no logo*, but I have never changed anything regarding the logo, I only flashed @Ralim's. firmware.

On the **TS80P** there *is a logo*.

paulfertser, 2020-09-11:
For me it also shows Sleeping... and starts to heat up to 150 C (which I set as sleeping temperature in the settings) and it continues to show Sleeping and to keep 150 C. That's in Auto S mode. To what temperature does it heat up for you?

discip, 2020-09-11:
I am mostly concerned about the **O** *mode*, since it is the mode I use.
And in this mode it should keep sleeping, until it is moved.

But the **S** *mode* does not work for me either, because it heats up even beyond the configured *sleep temperature*.

One more thing is, if the **S** *mode* is selected it should display ***`Sleeping ...`*** and heat up to the desired *sleep temperature* in the background.

paulfertser, 2020-09-12:
@discp, thank you for the clarifications. For me O mode doesn't start heating at all, and S mode heats up just to 150 C (which matches the settings), in both cases after power on I see "Sleeping..." for as long as I do not move the iron. I also have logo enabled which might probably make a difference (by adding a few seconds delay at the startup). And probably my TS100 vs. your TS80 is also playing a role. To sum up, unfortunately, I can't reproduce and thus can't really help, but I hope our conversations helped to clear up the details and probably @Ralim will now be able to see it.

discip, 2020-09-12:
@paulfertser 
Thank you for your investigation and willingness to help.

## Kudos for that. Please keep it that way. 👍

Ralim, 2020-09-14:
@discip 
It does sound like your iron is reading a higher vibration than the sensitivity level and therefore thinking that it _should_ exit the sleep modes.

Is this a regression compared to previous firmware versions? I might need to just delay the accelerometer readings to let them stabilise a bit more if so.. 😅

discip, 2020-09-14:
@Ralim 
In terms of stability and functionality it is definitely a regression, at least for the **TS80**, because this worked flawlessly after @firebie hast solved 'my *problem* (#571)'.
On the **TS80P** it did not work yet, so I think it is not correct to call it a regression.

### Thank you for your time. 👍

Firebie, 2020-09-14:
The idea was to fill up movement data array with initial accelerometer readings:
```cs
		datax[currentPointer] = (int32_t) tx;
		datay[currentPointer] = (int32_t) ty;
		dataz[currentPointer] = (int32_t) tz;

		if (!accelInit) {
			for (uint8_t i = currentPointer + 1; i < MOVFilter; i++) {
				datax[i] = (int32_t) tx;
				datay[i] = (int32_t) ty;
				dataz[i] = (int32_t) tz;
			}
			accelInit = 1;
		}
```

Firebie, 2020-09-14:
I think that issue is here (GUIThread.cpp:239, used 100 instead of 1000 or better pdMS_TO_TICKS(1 * 1000)):
```cs
if ((xTaskGetTickCount() > 100) && ((accelInit && (xTaskGetTickCount() - lastMovementTime < 100)) || (xTaskGetTickCount() - lastButtonTime < 100))) return 0;  // user moved or pressed a button, go back to soldering
```

discip, 2020-09-14:
@Firebie
I _somehow_ knew i could count on you. 😁
## Kudos for investigating on that! Highly appreciated! 👍
Is there any chance, of checking, if your assumption is right. 😉
I would love to test compiled versions of firmwares, which were alternated in the way you suggested. (*If only I knew, what exactly has to be changed!*) 😩

Firebie, 2020-09-14:
Try this: [TS80P_EN.hex.fix.1.zip](https://github.com/Ralim/ts100/files/5220427/TS80P_EN.hex.fix.1.zip)

discip, 2020-09-14:
First of all:
### Thank you!
But unfortunately this did not solve the problem. 😔

discip, 2020-09-14:
@paulfertser 
I know it is ***off topic***, but since this is not crucial, i did not want to open a new issue on that.
It is about the blinking frequency of the 'scroll bar':
To me it seems somewhat hectic. 😅
Would this take much effort to make it slower?

Firebie, 2020-09-14:
Try this: 
[TS80P_EN.hex.fix.2.zip](https://github.com/Ralim/ts100/files/5220835/TS80P_EN.hex.fix.2.zip)

discip, 2020-09-14:
No, this did not work either.
Now even the **`Sleeping...`** on power up is gone.

Firebie, 2020-09-14:
Ver 3: 
[TS80P_EN.hex.fix.3.zip](https://github.com/Ralim/ts100/files/5220984/TS80P_EN.hex.fix.3.zip)

discip, 2020-09-14:
Nope, sorry sir, but still not there.
But it shows now the **`Sleeping...`** again.

Firebie, 2020-09-14:
Just to understand - it shows that it sleeps, but heats up to normal soldering temp?

discip, 2020-09-14:
Yep! And all that, **without being moved!**

Thank you again for working on the solution for this!

Unfortunately I have to go to bed now. 🥱
So testing will be resumed tomorrow, if you are still willing.

Firebie, 2020-09-14:
When it shows sleep, it has to show also current tip temp.
Which tip temp it shows in your case?

discip, 2020-09-15:
@Firebie 
**Sorry**, unfortunately I have to **revise** my last answer from yesterday. 😔
It clearly was to late. Despite having read you post twice, I misunderstood your question.
The correct answer is:
It shows **`Sleeping...`** on being plugged in, but shortly after that, skips it and starts heating up (*without the **`Sleeping...`** indicator*).

The *tip temp* is **26°C**.

thanks & kind regards

Firebie, 2020-09-15:
Ver 4: 
[TS80P_EN.hex.fix.4.zip](https://github.com/Ralim/ts100/files/5223165/TS80P_EN.hex.fix.4.zip)

discip, 2020-09-15:
@Firebie
# You are is incredible!
For **O** mode this works at first glance.
Let me test it on different scenarios and then I will report back.

ps: Could you please provide the**TS80** build as well?

Thank you very much.

discip, 2020-09-15:
Sorry I celebrated to early, with the connection to my computer **5V** it worked, but for 12V PD it failed.

discip, 2020-09-15:
However at least **S** mode works now as intended. 👍

Firebie, 2020-09-15:
Ver 5 (TS80) - delay 2 seconds: 
[TS80_EN.hex.fix.5.zip](https://github.com/Ralim/ts100/files/5223330/TS80_EN.hex.fix.5.zip)

discip, 2020-09-15:
Refering to: **TS80P_EN.hex.fix.4.zip**

For **S** mode it only works, if:
**`current tip temp`** is greater than **`sleep temp`** or if
**`current tip temp`** is much smaller than **`sleep temp`**.

Firebie, 2020-09-15:
For sleep modes (stayOff - O mode) we have next code:
```cpp
			currentTempTargetDegC =
					stayOff ?
							0 :
							min(systemSettings.SleepTemp,
									systemSettings.SolderingTemp);
```

discip, 2020-09-15:
Sorry, I do not really know, what I am supposed to do with this information.
### What I can say though: **V5** for the **TS80** works as intended for both modes!

Firebie, 2020-09-15:
Ralim was right, we need delay before we start reading data from accelerometer.
In V5 it is 2 seconds.
V6 - no changes from code except delay for 2 seconds: [TS80_EN.hex.fix.6.zip](https://github.com/Ralim/ts100/files/5223443/TS80_EN.hex.fix.6.zip)

discip, 2020-09-15:
### 👍 This seems to work! For both *'S'-* & *'O' mode* & *TS80* & *TS80P*
The only thing I noticed while **`Sleeping...`** is active, the iron does delay the negotiation for **12V** as well, because it does that a bit later than usual.

Firebie, 2020-09-15:
Power check added earlier: 
[TS80P_EN.hex.fix.7.zip](https://github.com/Ralim/ts100/files/5223520/TS80P_EN.hex.fix.7.zip)

discip, 2020-09-15:
May I ask for the **TS80** build as well? 😊

Firebie, 2020-09-15:
Sure 
[TS80_EN.hex.fix.7.zip](https://github.com/Ralim/ts100/files/5223552/TS80_EN.hex.fix.7.zip)

discip, 2020-09-15:
Sorry for being *undecided*, but I think it is **actually even better**, to **delay** the **heat up *'speed'*** *(By delaying the power check.)* even after the triggering of the accelerometer.

Because when you pick the **'S'** mode you do not want to start soldering right away.
So it would be better in terms of **wear of the tip** (That is why I prefer **'O'** mode over **'S'** mode.) and in terms of **power draw** *(hence money)*.

### If I am wrong, I am very appreciative for being corrected. 😀

paulfertser, 2020-09-15:
> It is about the blinking frequency of the 'scroll bar':
> To me it seems somewhat hectic. 😅
> Would this take much effort to make it slower?

Not much effort but the code will get a bit more complicated and less
maintainable IMHO. But that's not something for me to judge.

Firebie, 2020-09-15:
[TS80P_EN.hex.fix.8.zip](https://github.com/Ralim/ts100/files/5223766/TS80P_EN.hex.fix.8.zip)
```diff
diff --git i/workspace/TS100/Core/Threads/MOVThread.cpp w/workspace/TS100/Core/Threads/MOVThread.cpp
index 4ef12c21..355af584 100644
--- i/workspace/TS100/Core/Threads/MOVThread.cpp
+++ w/workspace/TS100/Core/Threads/MOVThread.cpp
@@ -45,6 +45,12 @@ void startMOVTask(void const *argument __unused) {
 	postRToSInit();
 	OLED::setRotation(systemSettings.OrientationMode & 1);
 
+	if ((PCBVersion == 1
+		|| PCBVersion == 2)
+		&& (systemSettings.autoStartMode == 2
+		|| systemSettings.autoStartMode == 3))
+		osDelay(2000);
+
 	lastMovementTime = 0;
 	int16_t datax[MOVFilter] = { 0 };
 	int16_t datay[MOVFilter] = { 0 };
```

discip, 2020-09-15:
@Firebie
# Kudos for being so friendly & obliging!
@paulfertser
> But that's not something for me to judge.

Neither do I want to arrogate to be able to do so.
But, **if no one minds**, I would like to have it blink slower.
So, I am officially making a request for this to be implemented. 😁
Unless somebody has a reason for this not being done.

Since this is solved, I am closing this issue.

Firebie, 2020-09-15:
Change for delay for accelerometer has to be review/merged by Ralim first.

discip, 2020-09-15:
Yes, I am aware of this. 😊
And I am confidently looking forward to this being done.
Are you going to make a PR for this, or is this not necessary?

Firebie, 2020-09-15:
Ralim is invited here to review changes, if they are ok, I will create PR.

Ralim, 2020-09-15:
Thank you all,
@Firebie solved this before I even got back around to looking into this after work 😅 
@discip If you need, you can easily use the artefacts that are generated from the merge for other irons should you need to.
https://github.com/Ralim/ts100/actions

discip, 2020-09-15:
@Ralim 
Thank you.
@Firebie upgraded my knowledge regarding this topic on the last encounter of accelerometer issues.

discip, 2020-12-29:
Sorry, but the latest commit https://github.com/Ralim/ts100/commit/768d36c060ace4454ff058b9125ea1d3316afb1e requires this to be reopened again, because in both modes **`S`** and **`O`** the iron behaves, as if it was in **`T`** mode, meaning it starts heating as soon as it is powered up.

  Firmware Version|**[v2.12.768D36C](https://github.com/Ralim/ts100/commit/768d36c060ace4454ff058b9125ea1d3316afb1e)**
-|-
  PCB Version|**2**
  Power Supply|**[UGREEN 65W GaN Charger](https://www.ugreen.com/products/65w-gan-usb-c-fast-charger?variant=32127034654782)**

thanks in advance

Ralim, 2020-12-29:
The latest commit does not change anything to do with this?

discip, 2020-12-29:
@Ralim
Good evening sir,
I do not know, since when this reentered the stage, but I just wanted to make sure, I did not forget to mention, which version I am currently using.

Ralim, 2020-12-30:
Could you give a check to see if https://github.com/Ralim/ts100/commit/7f81fbbe7a3bea9b92d98343112a2e79b4adccd3 helps with this?

Ralim, 2020-12-30:
you can check the latest CI build to test this (currently [this one](https://github.com/Ralim/ts100/actions/runs/452337083)).

Just scroll down and click on your model to download a zip of all of the languages.
🙏🏼

discip, 2020-12-30:
@Ralim
First of all, thank you for the instant reaction.
Unfortunately this is not resolving the problem entirely, because the iron starts to heat up to soldering temperature, shortly after it seemingly behaves as expected, when in **`S`** or **`O`** mode.
Is the solution https://github.com/Ralim/ts100/pull/699 that was implemented by @Firebie still present?

thanks in advance

Ralim, 2020-12-30:
Yep that fix is still present.

Could you check if in the debug menu the mov time is increasing when the unit sits still?

discip, 2020-12-30:
No it does not.

Ralim, 2020-12-30:
Ctip and Chandle are different as im printing them in the resolution we are using internally for most of the code. Since its sorta debugging info i've just left them at defaults?

30 `uV` is around 1C. There is always a tiny amount of oscellation in the tip temperature, as its also sensitive to noise from the rest of the system. Generally 30uV is _fine_ as it will be filtered out later on anyway.

I believe this is fixed again in the lastest builds [here](https://github.com/Ralim/ts100/actions/runs/453539591)

closed 2020-12-30 by Ralim

discip, 2020-12-31:
@Ralim 
## Thank you very much that worked! 👍

Ralim, 2020-12-31:
Always happy to help, apologies for giving the run around for asking extra questions before i took a look.

[1 more reply]

commit 29863ebf (2020-12-30): Edit sleep mode settings to be autostart aware

Fixes #696