Ralim/IronOS #698: ts100 not working

mitko3, 2020-09-13:


paulfertser, 2020-09-13:
Hello @mitko3,
This firmware doesn't support any communication with the computer over USB at all, it's not needed since all the parameters are configurable with the on-screen-menu. Where did you buy your iron from, an official distributor? Or can it be a fake one? Do you see anything on screen at all when powering it up with this firmware?

Ralim, 2020-09-14:
@mitko3 

* Which firmware did you load to the unit (official or this one?)
* Are you sure you are booting into DFU mode (it will say DFU on the screen)
* What version is the DFU on your iron ? (It will show it on the screen)
* What model is your iron (is it an official one?)
* What computer are you using
* Are you sure you read all the of advise documented in this repo for the issues that are known with the DFU mode

mitko3, 2020-09-14:
-I tried several versions
-yes
-DFU 3.45
-my model sequre sq-001 mini 
-i use windows 10 ,ryzen 3

I tried everything but it doesn't work, nothing is displayed, and when I connect it to the computer without pressing anything, I only hear a sound that it is connected but it doesn't work, and when I took it, the config file came out

Cpt-Huck, 2020-10-03:
Hello :) 
seems like I have the same ( maybe the exact same ) problem

paulfertser, 2020-10-03:
Hey :)

> seems like I have the same ( maybe the exact same ) problem

I thought STM32 MCUs are so popular that it's not hard to find someone
familiar with them locally. The iron is very easy to disassemble and
the schematics of the previous (working) version are freely available,
so it's not a major RE effort to see what makes the new hw revision
different. Probably it's a slightly different display model?

Cpt-Huck, 2020-10-03:
...Mine was working.. checking from time to time if there are updates available
..saw there are newer versions, so I updated mine and my dads TS100.. 

on mine, the DFU wasn`t working correctly somehow..
( TS100 still working...)

tried a few times to get the update on it.. which didn`t worked.. and now I cannot solder anymore.. only DFU "works" - well.. not really.. read further..

when pressing the Button near to the TIP while powering on ( and release the button ) will enter DFU mode ( normal behavior )

when holding the Button near to the TIP makes the DFU Mode enable, disable, enable... and so on... ( in other words: start, stop, start... ) ( not normal behavior -> normally stays in DFU ) 

just plug in micro USB or DC input and the TS100 stays dark ( OLED ) 

DC input: pressing the Button near to the TIP won`t do anything.. Besides going into DFU -> Tip is not heating up.. 

I could try USB + DC Input at the same time, but i dont know if this is safe to do 🤔 

we have two TS100 at home, the other one updated without any problem.. 

both had the same?! Ralim FW on it.. I guess it was v2.10.1..maybe v2.09 or v2.08.1

I try`d switching the STM32 PCB from the working one and put it into the defective one.. It worked !! 

..tried the defective one in the one which was working and It doesn`t worked..

so it`s the STM32 PCB which is somehow "broken"..

I guess the FW got corrupt, but I cannot Update it because Windows doesn't let me access it ( see Pictures ) 
So the DFU on the TS100 doesn't work correctly.. 

but the STM32 PCB has four solder pads.. maybe it can be flashed manually ? ( see Pictures )

Which firmware did you load to the unit (official or this one?)                                - Ralim

Are you sure you are booting into DFU mode (it will say DFU on the screen)         - yes

What version is the DFU on your iron ? (It will show it on the screen)                     - both 3.45

What model is your iron (is it an official one?)                                                          - TS100 - my dad ( working one ) has PCB 1.. / bought mine in summer this year ( broken one ) 

What computer are you using                                                                                   - Win 10 - try`d other Win 10 also and diff USB Cables and USB Ports..

Are you sure you read all the of advise documented in this repo for the issues that are known with the DFU mode - I guess so 🤔 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

paulfertser, 2020-10-03:
> but the STM32 PCB has four solder pads.. maybe it can be flashed manually ? (
> see Pictures )

See
https://github.com/Ralim/ts100/blob/master/Documentation/Upgrading.md
, yes, they're for flashing and debugging if you have an SWD adapter
(Raspberry Pi can be used as one with OpenOCD). And you have newer
version 2.51A so you can also debug and flash with SWD via the USB
connector I guess.

The pictures you posted are interesting. It looks like some contacts
got solder on them (accidentally). I do not think it's possible to
clean it off anyhow but you can ensure that the tainted contacts are
springy (if they should be) and you can try cleaning them with alcohol
and see if it changes anything.

Cpt-Huck, 2020-10-04:
Thanks for linking the dokumentation ( didn`t thought about that )

But i still dont get it.. which pin is what ?!

Hope you can help..

didn`t find it here ( don`t really know how to read it )
https://github.com/Ralim/ts100/blob/master/Development%20Resources/TS100%20V2.46%20Schematics%20V1.0.pdf

For flashing, I have: 
- FTDI ( can be used for updating ISDT FW )
- ST-Link V2 ( STM8 & STM32 )
- Raspberry Pi 3 Model B

never really used them..  - only the FTDI for updating ISDT FW- and the Raspi for Octoprint 3D Printer SW

on the bottom of the MCU riser PCB stands "v2.51A"

- looks more like solder than corrosion or "over current ?!"( wasn`t me trying to solder LOL :D )
Already tried to clean it with Alcohol ( Isopropanol )( before taking the picture )

- can`t really tell if the contacts are springy..
need smaller / finer tools...

paulfertser, 2020-10-04:
> But i still dont get it.. which pin is what ?!

See here: https://github.com/wose/ts100

But probably it'd easier to connect via the USB pins, those I
documented in the Upgrade.md file, and no soldering required :)

>   • ST-Link V2 ( STM8 & STM32 )

This would be the easiest way.

>   • looks more like solder than corrosion or "over current ?!"( wasn`t me trying
>     to solder LOL :D )

Hard to tell from a photo. BTW, Miniware might send you a new MCU
board if you show the picture to them and ask what to do.

>     Already tried to clean it with Alcohol ( Isopropanol )( before taking the
>     picture )

Did it not change the observed behaviour anyhow?

Cpt-Huck, 2020-10-04:
> See here: https://github.com/wose/ts100

I guess that helped a lot !

> But probably it'd easier to connect via the USB pins, those I
documented in the Upgrade.md file, and no soldering required :)

I cannot acces ( sometimes for a Sek ) the TS100.. 
Write something on it doesn't work at all.. 

Repost of the Pictures here:
[image]
[image]

So the DFU mode is seems to be corrupt ?! 🤔
DFU = Bootloader  ? 

Need to take a look on it later this day.. 😅 
Thanks already for helping :+1: 

> BTW, Miniware might send you a new MCU
board if you show the picture to them and ask what to do

sales@e-design.com.cn

This email I guess... From the Miniware Website 

> Did it not change the observed behaviour anyhow

Nope :/

paulfertser, 2020-10-04:
>   But probably it'd easier to connect via the USB pins, those I
>   documented in the Upgrade.md file, and no soldering required :)
> 
> I cannot acces ( sometimes for a Sek ) the TS100..
> Write something on it doesn't work at all..

I'm talking about using SWD over the USB connector, as SWD lines are
bridged to USB data lines on this board revision.

> So the DFU mode is seems to be corrupt ?! 🤔
> DFU = Bootloader ?

Eh, well, there's a standard DFU protocol for USB updates (there's
another for BLE etc) but unfortunately the bootloader flashed by
Miniware is not using it (pretending to be a mass storage device is a
fragile and silly way to implement the update mechanism). dap-boot
bootloader implements proper DFU which is compatible with dfu-util and
works reliably.

The bootloader is very unlikely to be corrupt. I'd guess it's some
hardware issue, probably just bad connectivity with that damaged
connector.

>   BTW, Miniware might send you a new MCU
>   board if you show the picture to them and ask what to do
> 
> [4]sales@e-design.com.cn
> 
> This email I guess... From the Miniware Website

Worth a try. Not sure how to check if that's exactly the address the
others used to contact them with hardware issues.

Ybriknevets, 2020-10-10:
I appear to have the same problem. When I connect the iron in DFU mode it appears as a drive but when I try to copy the firmware over the drive immediately vanishes and the file transfer process exits with an unknown error. When I click try again it gives the error that the drive is too small. If I leave the iron connected some time later the drive shows up again but I get a prompt to insert a disk into it and I cannot see it's contents in explorer just a white window. Cpt-Huck did you ever manage to restore your iron's functionality?

Ybriknevets, 2020-10-10:
> I appear to have the same problem. When I connect the iron in DFU mode it appears as a drive but when I try to copy the firmware over the drive immediately vanishes and the file transfer process [...]

Issue solved simply by moving the iron to a different USB port.

digitalis-ts100, 2020-10-28:
I got my ts100 DFU 3.45 2.5B mobo. last firmware i flashed 2.10. My oled died on me so i bought a new one. all is ok.
Today i tried 2.12b. it flash OK. i got .RDY. but when power on i got "BLANK". thou sometimes it will show garbage on boot, then blank again. but using miniware 2.18 works. 
I was checking the OLED.C/OLED.H i saw ralim used SSD1307. while miniware SSD1306.
could it be the cause? Thanks

Ralim, 2020-11-01:
@digitalis-ts100 

No problem at all; I've certainly had some flaky oled soldering cause intermittent issues.
This firmware does run the screen "harder" in that it updates more and the I2C bus is a _lot_ busier; so in the Miniware firmware you can get away with floating signals a lot better :/

@Cpt-Huck 
Most likely you have bad connections causing it to drop off the USB bus causing that issue.
That connect definitely looks unhappy, was this iron missing its foam piece that holds in contact?

digitalis-ts100, 2020-11-02:
> @digitalis-ts100
> 
> No problem at all; I've certainly had some flaky oled soldering cause intermittent issues.
> This firmware does run the screen "harder" in that it updates more and the I2C bus is a _lot_ busier; so in the Miniware firmware you can get away with floating signals a lot better :/
> 
Aha. Yep. I knew the only repairs I did was. Changed a burnt out oled. 
Thanks for elaborating. 
I like to add that my psu connector was acting strange and I had to fixed it. 
Now my ts100 with 2.12b works great.

Cpt-Huck, 2021-01-06:
> [image]
> [image]
> 
> ...Mine was working.. checking from time to time if there are updates available
> ..saw there are newer versions, so I updated mine and my dads TS100..
> 
> My dads TS100 updated without any problem..
> 
> on mine, the DFU wasn`t working correctly somehow..
> ( TS100 still working...)
> 
> tried a few times to get the update on it.. which didn`t worked.. and now I cannot solder anymore.. only DFU "works" - well.. not really.. read further..
> 
> when pressing the Button near to the TIP while powering on ( and release the button ) will enter DFU mode ( normal behavior )
> 
> when holding the Button near to the TIP makes the DFU Mode enable, disable, enable... and so on... ( in other words: start, stop, start... ) ( not normal behavior -> normally stays in DFU )
> 
> just plug in micro USB or DC input and the TS100 stays dark ( OLED )
> 
> DC input: pressing the Button near to the TIP won`t do anything.. Besides going into DFU -> Tip is not heating up..
> 
> I could try USB + DC Input at the same time, but i dont know if this is safe to do 🤔
> 
> both had the same?! Ralim FW on it ( think so, not 100 % sure) ..
> I guess it was v2.10.1..maybe v2.09 or v2.08.1
> 
> I try`d switching the STM32 PCB from the working one and put it into the defective one.. It worked !!
> 
> ..tried the defective one in the one which was working and It doesn`t worked..
> 
> so it`s the STM32 PCB which is somehow "broken"..
> 
> I guess the FW got corrupt, but I cannot Update it because Windows doesn't let me access it ( see Pictures ) ( will be added later ) ⚠
> So the DFU on the TS100 doesn't work correctly..
> 
> but the STM32 PCB has four solder pads.. maybe it can be flashed manually ?
> ( see Pictures )
> 
> I also answered these questions:
> Which firmware did you load to the unit (official or this one?
> - Ralim
> 
> Are you sure you are booting into DFU mode (it will say DFU on the screen)
> - yes
> 
> What version is the DFU on your iron ? (It will show it on the screen
> - both 3.45
> 
> What model is your iron (is it an official one?) -
> - TS100 - my dad ( working one ) has PCB 1.. / bought mine in summer this year, maybe PCB 2 ?!? dont know.. ( broken one )
> 
> What computer are you
> - Win 10 - also try`d other Win 10 PC and diff USB Cables and USB Ports..
> 
> Are you sure you read all the of advise documented in this repo for the issues that are known with the DFU mode - I guess so 🤔
> 
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]

I now know what`s causing the Problem: the STM32 Microprocessor has loose contact (because of the solder on the contacts i guess) I also did a (4 min) Video about it, in which I go more into detail..: https://youtu.be/DmbiolnIGoY

discip, 2022-04-23:
@mitko3
Were you able to solve this issue?
If so, please consider closing it. 😊

thanks

dickysoe, 2022-06-04:
I have the same problem with my sequre sq-001.
were you guys able to get the soldering iron working again?

thanks

dickysoe, 2022-06-06:
> 

turns out my SQ-001 got CKS32F103T8U6 chip. not the original STM32F103 chip. 
probably why i cannot use ST-link utility to load the firmware. chip ID was not recognized by the utility.

Ralim, 2022-06-06:
Ah a clone would certainly explain it. You can use the st-link itself with other tools but not the st-link util.
This firmware may or may not work on that clone device; I have no idea how closely they have copied the ST design sadly

dickysoe, 2022-06-07:
> Ah a clone would certainly explain it. You can use the st-link itself with other tools but not the st-link util.
> This firmware may or may not work on that clone device; I have no idea how closely they have copied the ST design sadly

Thanks for replying. I have 2 of sq-001. The first one is using you ironOS, and not problem at all. Working great.
The last one, with CKS chip failed during flashing.
I'm still trying to get the OS using SWD & stlink v2. But still cannot get it to work. Chip or coreID probably the problem.

yilmazbozaci79, 2022-11-16:
Hello, can anyone solve the Ts100 soldering iron update problem? The CONGIF file does not appear, the version does not appear, and the TS 100 does not open. an unexpected error while copying the update file prevents you from copying the file gives an error that a non-existent device was specified! Can you help me in this matter?

Jonny-T-cyber, 2022-11-29:
> -I tried several versions -yes -DFU 3.45 -my model sequre sq-001 mini -i use windows 10 ,ryzen 3
>
> I tried everything but it doesn't work, nothing is displayed, and when I connect it to the computer [...]

I follow.
I have the same problem...
I connect the soldering iron to the pc but it is not recognized by windows and no new drive appears in which to insert the firmware to be flashed

CoughCool, 2023-07-22:
> solved simply by moving the iron to a differ

Had the exact same issue on Windows 11. Moved my USB cable connected to the T100 to one of the rear USB ports and I was able to copy the .hex file to the iron.

sanekkk777, 2023-09-21:
> Ah a clone would certainly explain it. You can use the st-link itself with other tools but not the st-link util. This firmware may or may not work on that clone device; I have no idea how closely they have copied the ST design sadly

Bro... maybe for you or all... I just found a solution (for myself) and on my version of Windows
Windows 10 PRO 22H2
Soldering iron TS 100 B2 \ DFU 3.49 \ SW 2.20 (stock)

I went through all the stages, and the download takes place the second time (the first file is ERR, the second (immediately, without remounting) RDY)

I only changed:
My computer -> Device Manager -> Disk devices -> Your soldering iron -> RMB properties -> Policy -> Disable buffer clearing <...>
My computer -> Device Manager -> Disk devices -> Your soldering iron -> RMB properties -> Volumes -> Fill

Check!