Ralim/IronOS #701: TS80P handle temperature often reading as 0

tjader0, 2020-10-05:
I'm using 2.12 beta on my TS80P and often my handle temperature is showing up as 0 °C in the debug menu.

After some investigation it seems that for the TS80P it's using a table lookup instead of a formula, and the table is only filled between 0 °C and 29 °C:

https://github.com/Ralim/ts100/blob/138000f4a5cfba7b6f1f0775cd6f67a2c764666d/workspace/TS100/Core/BSP/Miniware/BSP.cpp#L23-L86

Where I live ambient temperatures above 29 °C are not uncommon, and in many places below 0 is also not uncommon.

It seems the plan is to eventually move to a formula, and I assume the table is small because of size restrictions, but maybe it would be saner to `return NTCHandleLookup[((i - 1) * 2) + 1] * 10;` instead of `return 0` when you don't find a proper value in the table, so that you return the maximum value, which should be closer to the real temperature than 0.

Another thing that could be saner without needing more space is to have the table move 2 °C per entry, then you could have something like from -15 °C to +43 °C in the table with 2 °C precision.

Ralim, 2020-12-30:
Hia,

I _believe_ this should now be fixed.
If you get some time, it would be fantastic if you check the latest CI build to test this (currently [this one](https://github.com/Ralim/ts100/actions/runs/452337083)).

Just scroll down and click on your model to download a zip of all of the languages.
🙏🏼

tjaderxyz, 2020-12-30:
Thanks!

I am currently traveling, but I will test it when I get back home in a couple of weeks.

tjaderxyz, 2021-01-11:
I just tested again on 2.12-beta and on 2.13. On 2.12-beta it was reporting CHan as 0, 2.13 reports it as 340, so it seems it has been fixed indeed.

https://github.com/Ralim/IronOS/commit/3696e08dde05d84be3769bb24382cf2a068e9262 this is probably the commit that did it.

closed 2021-01-11 by tjaderxyz