Ralim/IronOS #708: Custom logo not working

fender2469, 2020-10-16:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug/Feature Request/Question/Complaint]_

* **I have**
  - [ x] Searched previous issues 
  - [ ] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [x ] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
I updated to the latest firmware and then loaded my custom logo. Everything on the computer said .RDY. but when I plug in my iron, no logo and no menu items, just DC or anything on the right of the screen as you scroll through the menu items. What could be wrong? 

* **What is the expected behavior?**

***Steps to reproduce the bug:***

- Turn on...
- Hold....

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.x?
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

Ralim, 2020-10-20:
Hello,
It is quite possible that you have generated your logo with an old version of the converted which used an incorrect location in the hex file.

If you regenerate your logo file with the python tool you should not run into this issue anymore.

fender2469, 2020-10-20:
I will try that, thank you!

ZiGGstern, 2020-10-20:
Did you get it to work with the Python tool? Just want to know before i set up a python env just to make the hex. :)

fender2469, 2020-10-20:
I haven't yet. I will try to give it a go tonight.

fender2469, 2020-10-21:
I tried to understand using the python tool but I can't do script.  ugh, I wish I could.  Sorry, I really wish I could get this to work.

fender2469, 2020-10-21:
> Hello,
> It is quite possible that you have generated your logo with an old version of the converted which used an incorrect location in the hex file.
> 
> If you regenerate your logo file with the python tool you should not run into this issue anymore.

Hi, I tried to understand using the python tool but I do not understand script.  Do you have an updated version of the converter tool maybe. The one I used was v1.15  from 2017.

Ralim, 2020-10-21:
I haven't updated the converter tool since the python tool has completely replaced it and gone further in terms of features.
If you attach your image or email it to [me](mailto:ralim@ralimtek.com) I can convert it you.

ZiGGstern, 2020-10-21:
Thanks for the offer but I actually got it working late last night :)

fender2469, 2020-10-21:
Ralim, I email you the logo I would like converted. I really appreciate it.

Ralim, 2020-11-01:
@fender2469 I believe I sent back the processed logo's to you. 
Going to close this issue as I believe there is no issue here to be solved?

closed 2020-11-01 by Ralim

fender2469, 2020-11-01:
Yes you did. I'm sorry I didn't get back to you. Been busy. They work
perfectly. Thanks again!