Ralim/IronOS #711: No tip detected displays crazy temp.

GreekGeorge, 2020-10-30:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug/Feature Request/Question/Complaint]_

* **I have**
  - [X] Searched previous issues 
  - [ ] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [ ] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
When there is no tip, it shows a temp exceeding 450°C. Once the tip is reinserted it shows the proper temp.
* **What is the expected behavior?**
Should either display a message along the lines of "no tip detected" or simply not show a temperature.
***Steps to reproduce the bug:***
- Remove tip from iron
- Look at display (pic related)
* **What is the motivation / use case for changing the behavior?**
Doesn't really make sense.

* **What are you running:**
  - Firmware Version: 2.12
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 19V @ 3.42A Max.
[image]

Ralim, 2020-10-30:
Ah yes,

There used to be a tip removed string that would be printed in this case, I can't remember if it was removed or if it has just been broken.

Shall have a look see for you soon.

Firebie, 2020-10-30:
In a case of detailed view it shows 'Tip disconnected', in a case of classic view - temp.

Ralim, 2020-12-30:
Hia,

I _believe_ this should now be fixed.
If you get some time, it would be fantastic if you check the latest CI build to test this (currently [this one](https://github.com/Ralim/ts100/actions/runs/452337083)).

Just scroll down and click on your model to download a zip of all of the languages.
🙏🏼 

I fixed this by in classic view just removing the iron for starting to solder.
While its not _ideal_ its a bit more intuitive and im open to other ideas

GreekGeorge, 2020-12-30:
Cool! Thanks for working on the issue!
I'll check it out when I get home, and I'll let you know.

GreekGeorge, 2020-12-30:
@Ralim be advised, the new flash doesn't display anything other than settings. Is this the function that you're after?
Pic related:
[image]

Ralim, 2020-12-30:
Yep that's what I was thinking.

Open to other ideas of what to do, but felt that may be simplest ?

GreekGeorge, 2020-12-30:
Oh, I thought it was a bug! lol
Maybe a simple message like "No Tip" or "NaN" or something I think would be nicer, and would fill the empty space in front of the settings symbol. Iron version maybe?

Ralim, 2020-12-30:
Ask and you shall recieve 🤣 

Added an icon for no tip instead, is this nicer?
[here](https://github.com/Ralim/ts100/actions/runs/453592415)

iLLiac4, 2020-12-31:
Tip disconnect detection does not work if the tip is removed during the soldering. It does work only when the iron is not heating up.

Ralim, 2020-12-31:
Correct, that is normal as we don't want to kick user back to the main screen once you are in soldering mode.

GreekGeorge, 2020-12-31:
Hey, that's pretty good!
Thank you for fixing the issue.
Greetings from Greece, and have a happy new year!
[image]

closed 2020-12-31 by GreekGeorge