Ralim/IronOS #714: How to open TS80p?

Makhaira, 2020-11-06:
I was about to start playing with this firmware but my screen just died. There's some helpful stuff on here about sourcing replacements, but for the life of me I can't get the TS80p open. So far I've removed the little allen screw and tried gently prising up the panel around the screen without success - any tips please?

GreekGeorge, 2020-11-06:
If it's anything like the TS80, there's a hidden screw under the button cover.
Please check this video:
https://youtu.be/83STpUXHilk?t=232

Makhaira, 2020-11-06:
Hero thank you

hwater, 2020-12-10:
thanks also for this guide - my display getting darker and darker. i have seen, the oled inside gets dry and transparent at the corners

whitehoose, 2020-12-10:
If it's an 80p presumably it's still under warranty - why not contact your seller?

hwater, 2020-12-10:
no it is an TS80

whitehoose, 2020-12-10:
I was aiming at the OP - but good to know :)

closed 2020-12-13 by Ralim