Ralim/IronOS #715: I burnt my TS100 II (from https://github.com/Ralim/ts100/issues/649)

chuckyto, 2020-11-09:
After almost setting my TS100 on fire by connecting it to 48VDC, it didn't show anything on screen neither via USB nor DC.I have replaced a few components SO FAR.
D1 was fried.
R30 also cooked.
D3 burnt aswell.

Right now, the little thing works and heats the tip, but there are some rather big issues:
-as soon as it is connected, the tip starts heating up (no matter if configured not to do it)
-it heats up until the tip is red hot. I mean RED HOT, but it shows tip temperature 0 on screen. I disconnected if super fast when I noticed.

Accelerometer works fine, changing text orientation when configured.

I *think* I have already tested what can be wrong and right now am stuck. Any idea will be much appreciated.

Ralim, 2020-11-09:
Almost 100% chance you have blown the main MOSFET on the board and the op-amp used for Temperature sensing .

chuckyto, 2020-11-10:
Okay. I checked them and the mosfet appears to be shortcircuited. I have ordered the mosfet and the temperature sensor(just in case) and we will see in a few weeks how it goes. If this is no solution, I think I will better order a new one and use the older as a replacements source :/.
Thank you!

Ralim, 2020-11-15:
No problem at all.
I would advise to just test all of the mosfet/transistors for shorts :) 
Really, I would sanity check all components for shorts if you have the time.

closed 2020-11-15 by Ralim