Ralim/IronOS #716: TS100 QUERY

greverfbrayer, 2020-11-13:
[image]
It is normal that when connecting the ts100 to the computer it does not recognize it.
There should be no "CONFIG" note.
[image]
Or is it due to the latest update I upload?

Ralim, 2020-11-13:
If you are using this firmware (not the official firmware). 

This is entirely normal as noted in the readme there is NO USB in normal operation.

greverfbrayer, 2020-11-13:
@Ralim I was scared, I thought I blew it. Thank you. I am now calmer.

closed 2020-11-13 by Ralim