Ralim/IronOS #722: Pinecil is available but....

iLLiac4, 2020-11-24:
Here is https://pine64.com/product-category/pinecil/?v=fc0902513eb2 already available.
It says 24USD from 36USD but at the reality at the end it will cost around 42USD what is more then the TS100 can be obtained.

Firebie, 2020-11-24:
I have a question related to schematic - is not zener diode to have limiting resistor?
And second question - I see that used BZT52B20S - it is 20V, but specification is saying that supply voltage can be up to 24V - is not it too much for this small zener?

gamelaster, 2020-11-26:
@iLLiac4 it will stay 25USD, the price "35USD" is retail price, but since PINE Store is community store, the price is not retail, so 25 USD.

OndroNR, 2020-11-27:
@Firebie according to community chat communication, zener diode along with hall sensor are not actually populated in production batch (and some resistor has different value). Schematics will be updated after the batch is shipped. But there might be an issue as VBUS pin of FUSB302 max recommended voltage is 21V, absolute max voltage is 28V. Datasheet also says "extended exposure to stresses above the recommended operating conditions may affect device reliability." and VBUS pin should have (external) OVP. Not an issue according to Pinecil engineer.

OndroNR, 2020-12-02:
@Firebie Wiki page got updated with real production schematics and note "Recommend operating voltage 12-21V, maximum rating at 24V". Still not an issue they say, and no investigation or revision planned. Community discussion available on #pinecil channel at discord or another bridged chat service.

iLLiac4, 2020-12-02:
Hmmm. Power supplies are usually 12v, 24v so with 24v we are already on the limit? I can already smell burned mosfets and see candle lights :)

Ralim, 2020-12-02:
Hmmm 🤔

I'm not in the chat system usually so missed this chat over there.

Have been running mine at 24V occasionally on and off with no issue so far.

It's too late to change the first set of units I believe but can see about adding in some protection if it can fit in there.

OndroNR, 2020-12-02:
I believe worst case scenario over longer time on 24V is unreliable/dead USB PD support and DC only operation, no fireworks or dead mosfets.

@Ralim preproduction Pinecils should have 20V zener diode still populated. Is that zener diode still operational after directly powered by 24V? Just curious :)

Firebie, 2020-12-02:
Original TS100 has the same zener, but with additional resistor in line.

iLLiac4, 2020-12-03:
My ts100 is always connected to 24v power supply that also feeds variable power supply, so it is universal power supply for more than two, three years and it works without problems.
If this is true about failing (even if only PD) pinecil with 24V then this is a case for a failed design and should be fixed before any further production and publish the design (tutorial) so at least some users will be able to fixed their own pinecil that has already been sent. 
Mine will be powered from 24V permanently and if it fails than I will be covered with warranty since they were advertised as 24v.

OndroNR, 2020-12-03:
As pointed out in chat, items sold at pine64.com are covered only with 30 days warranty https://pine64.com/warranty-policy/

gamelaster, 2020-12-03:
@OndroNR yes, the reason why is explained here: https://www.pine64.org/2020/12/02/pine-store-community-pricing-online-retail-stores/

iLLiac4, 2020-12-03:
Hmmmm. This is then stupid because there is no warranty then, considering that already the delivery is taking so long.

MstrVLT, 2020-12-29:
[image]
D1 fell out of the hole from tip side. Soldering iron works wel. I don't know what to do with D1

OndroNR, 2020-12-29:
Was it D1 for sure? D1 zener diode should not be populated on production Pinecils. Maybe you accidentally received preproduction sample (already happened at least once according to Discord discussion), unpopulated D1 is not only change. I recommend contacting Pine64 support.

Firebie, 2021-01-22:
I have a set of original miniware tips - almost all of them can't be inserted into pinecil - stuck in the middle.
Is it my version of pinecil so specific or it is all pinecils like this?

Ralim, 2021-01-23:
Check that the screws in top/bottom are not blocking inserting the tips.

Some of them are not perfect in their sizing and the rings are large enough to just clip on the screws

Firebie, 2021-01-23:
I even disassembled the unit and removed tip contacts - but it doesn't help. Miniware tips are suck at this position:
[image]

iLLiac4, 2021-01-23:
Yes the are not the same size. I doubt that it is intentionally. Original one is 5,73mm, the include one is 5,56mm.
So you can widen the hole in the insert a little bit and it will fit or. you can do that on the original tip and narrow it a little bit.

Ralim, 2021-01-23:
Hmm, I doubt it is intentional, they were aiming for compatibility.

I agree widening would work if your comfortable with it, but if not you might be able to talk to pine about it. Not entirely sure what they can do though

iLLiac4, 2021-01-23:
It is really easy to fix. Roll a small piece of grinding paper and insert it. Roll a few times and try. Repeat until if fits.

Firebie, 2021-01-23:
> It is really easy to fix. Roll a small piece of grinding paper and insert it. Roll a few times and try. Repeat until if fits.
Thanks, helped!

closed 2021-04-05 by Ralim

Conor-Burns, 2021-06-06:
Hi sry for necro - but i think I fried the USB PPD IC with 24v so there's that :/

Conor-Burns, 2021-06-06:
Going to order a FUSB302MPX chip and replace it to see if it's really that one who died

Firebie, 2021-06-25:
Is it only me with the Pinecill which stuck on load?
If I comment out these 2 lines, no stuck happends:
```c++
  /* definition and creation of POWTask - Power management for QC */
  osThreadStaticDef(POWTask, startPOWTask, osPriorityAboveNormal, 0, POWTaskStackSize, POWTaskBuffer, &POWTaskControlBlock);
  POWTaskHandle = osThreadCreate(osThread(POWTask), NULL);
```

Ralim, 2021-06-25:
@Firebie 
Please raise a separate issue for this. This issue is closed