Ralim/IronOS #724: TS100 strange sound under OLED screen

SuperNeko, 2020-12-01:
_This is a [Question]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

On firmware 2.11, when iron is turned on and heated up, I can hear a clicking sound (like clock, but faster) under OLED display.
On firmware 2.05 I can hear the same sound, but it is more like buzzing(maybe higher frequency).

***Steps to reproduce the behavior:***

- Turn on...
- Heat it up and listen closely to OLED display.

* **What are you running:**

  - Firmware Version: 2.11.372F8E3
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 24V 3A 

Looking up at Google, I only found similar behavior here: 
https://github.com/Ralim/ts100/issues/628

So my question is: is it normal behavior or something is wrong with my soldering iron?

Ralim, 2020-12-01:
This is normal.

It's because of the capacitor that ac couples the PWM drive signal.
This firmare takes the tradeoff of running that slightly more into the audible range but this provides slightly better turn on of the pass fet and therefore better power to the tip.

SuperNeko, 2020-12-01:
Thanks for your answer, I really appreciate that. Issue solved.

closed 2020-12-01 by SuperNeko