Ralim/IronOS #728: Problem with display polish "ł" letter.

canislupus11, 2020-12-07:
There is a problem with the display of the letter "ł" when using the Polish language version. Instead of "w" all you see is a blank space.
Temporarily, for readability, in the translation corrections I have removed all the letters "ł" replacing them with "l". After fixing the bug, I will correct the translation.

  - Firmware Version: 2.11

Ralim, 2020-12-07:
Hmm, interesting.
All of the font data is generated from here: https://github.com/Ralim/ts100/blob/master/Translation%20Editor/fontTables.py
And it looks like that symbol _should_ be there :/

Ill have a look into this in a bit :)

KITT3000, 2020-12-07:
@Ralim The TS80 displays correctly, he has the TS100 and apparently that's the only problem there.

canislupus11, 2020-12-07:
W dniu 07.12.2020 o 19:32, Ziuta pisze:

canislupus11, 2020-12-07:
on my TS100: https://imgur.com/a/JDPda7z

Ralim, 2020-12-13:
@KITT3000 
Its the same code for displaying characters for all systems.
So TS80/TS100/TS80P/Pinecil all use the same font tables generated from the python code base.

@canislupus11 Are you using a local build of the firmware or the one from CI?

canislupus11, 2020-12-13:
> @canislupus11 Are you using a local build of the firmware or the one from CI?

I using firmware version downloaded from github release...

KITT3000, 2020-12-15:
@Ralim I downloaded the current pre-release today and there is also a lack with this letter on the TS80.

Ralim, 2020-12-17:
Interesting @KITT3000 
Thank you for that, helps narrow down the issue.

Can you confirm for me what the difference between `Ł` and `ł` is; and if to you this issue is only present for one or the other? (Not entirely trusting my ability to pick the difference as most of my tooling is matching both)

KITT3000, 2020-12-17:
@Ralim
 `Ł` - capital letter (used in a word that begins a sentence or a word as a proper name)
`ł` - lowercase letter

Ralim, 2021-01-02:
This should be fixed now.

canislupus11, 2021-01-14:
> This should be fixed now.

Seems to be OK. Thanks!

closed 2021-01-14 by canislupus11