Ralim/IronOS #731: Settings are unusable on Pinecil

jlevy-dev, 2020-12-10:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug/Feature Request/Question/Complaint]_

* **I have**
  - [ x] Searched previous issues 
  - [ x] This is in _this_ firmware, not vendor "offical" firmware
  - [ x] This is not a bug in the vendor bootloader (aka DFU)
  - [ x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
When I change the sleep timer settings I am not able to turn them back on, it goes to off with no other adjustments
* **What is the expected behavior?**
Be able to turn on the setting from off and adjust this setting
***Steps to reproduce the bug:***
[Video](https://photos.app.goo.gl/8f6rajM82eEeQyGQA)
- Turn on
-Tab over to the sleep settings
-Change time for it to take to sleep and it does not let me to change it or turn it back on unless I factory reset

* **What is the motivation / use case for changing the behavior?**
Not able to use baked in parts of this firmware

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

    Firmware Version: 2.12.a6edc9b
    PCB Version: 1.0
    Power Supply (Voltage and Current Rating) : N/A

Ralim, 2020-12-10:
Hello,
Are you sure thats the PCB version from the debug screen? We would be expecting `3` for the `BMA223` ?

jlevy-dev, 2020-12-10:
oh ya sorry pcb version 3.

Ralim, 2020-12-10:
Hmm. Interesting.
Will poke when I get a chance (Work life balance is a mess atm)

Ralim, 2020-12-10:
Yep its a dumb error I missed in testing. Will have it patched when I can get to windows machine to build pinecil #730

jlevy-dev, 2020-12-10:
I can build it if you need me too, did you push the code?

jlevy-dev, 2020-12-10:
Alright, thank you :)

commit 860c6714 (2020-12-13): Bufix: patch PCBVersion in accelerometer model

Closes #731
closed 2020-12-13 by Ralim