Ralim/IronOS #733: TS80 unable to mount after firmware update

mmehrle, 2020-12-12:
I can't remember the firmware I installed - it was probably three months or so ago. Anyway, I am experiencing strange issues in that the iron is unable to heat up, suddenly stops heating in the middle of soldering, etc. This is happening with my 30W USB QC 3.0 as well as a 6S 1800mAh battery I connected via an adapter (worked before just fine).

So I thought about a firmware update but to my dismay I am unable to actually mount it on both Windows and OS X. Not sure if that has to do with the firmware but I thought I'd reach out. At this point it's unusable as it's unreliable plus I am unable to update the firmware. Is it dead?

paulfertser, 2020-12-12:
Hi,

On Sat, Dec 12, 2020 at 06:59:23AM -0800, mmehrle wrote:
> So I thought about a firmware update but to my dismay I am unable to actually
> mount it on both Windows and OS X. Not sure if that has to do with the firmware

For firmware update you're supposed to press the button before
powering on the device so that the bootloader doesn't boot the
firmware at all.

This firmware is not mountable so if you're booting it no wonder you
can't mount anything.

-- 
Be free, use free (http://www.gnu.org/philosophy/free-sw.html) software!
mailto:fercerpav@gmail.com

paulfertser, 2020-12-12:
> up, suddenly stops heating in the middle of soldering, etc. This is happening
> with my 30W USB QC 3.0 as well as a 6S 1800mAh battery I connected via an
> adapter (worked before just fine).

This might probably be explained by poor contact between the USB
socket and the PCB. It's probably the most fragile point
mechanics-wise. I suggest to disassemble the device, insert a USB
cable and while looking (probably with magnification) at the USB
socket pads wiggle it a bit to see if there're any cracks.

mmehrle, 2020-12-12:
> Hi,
> On Sat, Dec 12, 2020 at 06:59:23AM -0800, mmehrle wrote: So I thought about a firmware update but to my dismay I am unable to actually mount it on both Windows and OS X. Not sure if that has to [...]

HA - I had totally forgotten about that. THANKS!!!

mmehrle, 2020-12-12:
Okay so I dropped the hex file on the drive and it disconnected. On relaunching it (had to do it manually) there's nothing on the drive. I assume that means the update was successful?

mmehrle, 2020-12-12:
Update - after the firmware update I tried it again but it's still acting quirky. Yeah I could take it apart but at this point I may just buy myself a TS80P.

Ralim, 2020-12-14:
Generally for QC if you get any loose connections on the data pins the supply device will cut the voltage to 5V for safety, which if occurs under load can sometimes undershoot the supply voltage.

I would first be suspect of trying a different cable.
Also you can try just injecting 9/12V directly down the usb wiring (down where 5V normally is) and this can bypass QC entirely. This will let you figure out if its the power contacts or if its the data pins.

closed 2021-01-16 by Ralim