Ralim/IronOS #736: Add PD status to the debug menu

Ralim, 2020-12-17:
This is a todo item im spinning out of #734.

Want to add some more metrics to the debug menu to indicate:
- FUSB present
- What state it is in

Civil, 2020-12-26:
So far I've for myself implemented simple debug menu which just prints out Status register statuses during initialization at various points (as dec10 values) and all status (status0, status1, status0a, status1a) on a screen which was helpful enough for me to look through datasheet and confirm that it's likely CC1 is in the air on my device (though I'm waiting for a replacement Pinecil, so I don't have a fully working device yet).

If you are interested in something like that, I can create a PR (or modify debug menu to include information in some more readable state)

Ralim, 2022-02-15:
Not sure how I missed this. 
Would be open to any ideas on how best to indicate the extra information :)

Civil, 2022-02-15:
M... I no longer remember what was the description of status registers in datasheet, but even raw dump + datasheet was already helpful for me to confirm that it was a hw failure.

So I guess it would be great to have human readable description of status registers of fusb302

closed 2023-07-21 by Ralim