Ralim/IronOS #739: [Question]Setting default temperature

radimsejk, 2020-12-19:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Question]_

* **I have**
  - [*] Searched previous issues 
  - [*] This is in _this_ firmware, not vendor "offical" firmware
  - [*] This is not a bug in the vendor bootloader (aka DFU)
  - [*] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
When I turn on the iron (I have set it so it will heat up automatically) it heats up to 300°C and I do not see the "default temperature" in settings.
* **What is the expected behavior?**
I would like to set my iron to heat up to 270°C automatically as this is my most frequently used temp.

* **What are you running:**
v2.12.
PCB 1
20V 3A power supply

Ralim, 2020-12-19:
Hi,

Just set the desired temperature and then exit soldering mode (hold the rear button or press both at once). When you exit back to the start page; it will save.

radimsejk, 2020-12-19:
Thank you, it helped me :)

closed 2020-12-19 by radimsejk