Ralim/IronOS #743: Temperature Calibration not working properly

Mortikhan, 2020-12-26:
This is a Bug

* **I have**
  - [X] Searched previous issues 
  - [X] This is in _this_ firmware, not vendor "offical" firmware
  - [X] This is not a bug in the vendor bootloader (aka DFU)
  - [X] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
Stock firmware (2.18) shows ambient room temperature(cool tip) as 72 F. 
2.12 Ralim Firmware shows cool tip temp at ~120 F
2.11 Ralim Firmware also shows cool tip temp at ~120 F
This after factory reset and temperature calibration (iron is cold hasn't been powered on today)

* **What is the expected behavior?**
Temp reading should reflect ambient temperature, or at least close to it. +-10F

***Steps to reproduce the bug:***
Install stock firmware - factory reset - reading is correct +_ 5 degrees
Install 2.12/211 Ralim firmware - factory reset - temp calibration - reading is +~40 degrees F above ambient
temp calibration shows 790-826 after pressing button, Idle temp then shows ~120 F

* **What is the motivation / use case for changing the behavior?**
+- 40 degrees is way out of calibration

* **What are you running:**
  - Firmware Version: 2.12.D75F361 / 2.11.372F8E3
  - PCB Version: (1/2) ?
  - Power Supply (Voltage and Current Rating) :12.20v 2.0A

Ralim, 2020-12-28:
Hi,

So the firmware runs internally in degrees C; and Fahrenheit support is added in by just converting to/from C.

Most likely this is just a loss of precision in `TipThermoModel::convertCtoF` and `TipThermoModel::convertFtoC` if I'm understanding the issue correctly?

Mortikhan, 2020-12-28:
It's either a loss of precision, or it is a flaw in the temperature conversion math.
C measurements are accurate, F Measurements are off by a constant amount. In my case ~25-30-ish degrees.
The ambient cool tip temp is off by the same amount the hot tip temp is off (for F)

Ralim, 2020-12-30:
Hia,

I _believe_ this should now be fixed.
If you get some time, it would be fantastic if you check the latest CI build to test this (currently [this one](https://github.com/Ralim/ts100/actions/runs/452337083)).

Just scroll down and click on your model to download a zip of all of the languages.
🙏🏼

Mortikhan, 2020-12-31:
Initial testing is positive.
The tip temp now matches the set soldering temp. 
I have to let the iron cool completely in order to test ambient temp readings.

Mortikhan, 2020-12-31:
Ambient temp is now within 5 degrees of ambient. 
Mine is currently reading 73 which is about 2 degrees high, but that is well within tolerances.
I'd say this is fixed.

Ralim, 2020-12-31:
🎉 
Fantastic.
Generally its +- a few deg C, off memory our references work out to about 1-2% accuracy.

closed 2020-12-31 by Ralim