Ralim/IronOS #748: Not an Issue, Looking for a manual.

Doyle4, 2020-12-29:
Has any one got a link to a manual please? New to the soldering iron and installed the firmware soon as I got it today.

PID tuned, Voltage tuned and set it how id like it, but wondering what Gain is and how I can tell what gain goes to what tip.

Thanks :)

Ralim, 2020-12-29:
Hello 👋🏼 

There is no "manual" for this firmware as a complete set.
There are docs in [here](https://github.com/Ralim/ts100/tree/master/Documentation)
Around things that differ from the main firmware from Miniware but also the iron has descriptions on all of the settings, if you leave it paused on them for a bit. (Trying to put documentation in the firmware :) )

Generally leave everything in Advanced options alone unless you want to tweak it :)

Doyle4, 2020-12-29:
@Ralim 
Thank you very much! So far so good, not used stock firmware, but your firmware was real nice to get around from the get go, thank you.

closed 2020-12-29 by Doyle4