Ralim/IronOS #749: TS80P v2.13 issue

Airmonkdev, 2020-12-31:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug]_

* **I have**
  - [ *] Searched previous issues 
  - [ *] This is in _this_ firmware, not vendor "offical" firmware
  - [ *] This is not a bug in the vendor bootloader (aka DFU)
  - [ *] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
Just flasher the new v2.13 on my TS80P but it seems that I can't enable the auto sleep and auto shutdown features.
Also (minor), powering the iron pushing the second button it show v2.12 31/12/20 instead of v2.13

* **What is the expected behavior?**
I was expect to enable both the automatic sleep and shutdown features.
 
***Steps to reproduce the bug:***
Goes into settings and try to enable the auto sleep or shutdown feature.
 
- Turn on...
- Hold....

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**
v2.13 on TS80P

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.13?
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

Ralim, 2020-12-31:
Ah bugger, need to fix that version string.

What PCB model is indicated in the debug menu for you? (You didn't put it in your query)

Airmonkdev, 2020-12-31:
Sorry, how can I enter debug menu and check it out?

Thanks for all you've done! ;)

discip, 2020-12-31:
@Ralim 
Unfortunately I have to report a new issue as well, but other than **OP**.

* The iron only occasionally negotiates for **PD**, for the most part it goes for **QC** (However nothing like that with version: https://github.com/Ralim/IronOS/commit/b359e6e37e9c10a0e21c89eb67cd20184dbb7bd3.)

iron|Firmware Version|PCB Version|Power Supply
-|-|-|-
**TS80P**|**https://github.com/Ralim/IronOS/commit/8acfc7e587184d947c054d7e039961b10a3c3e0e**|**2**|**[UGREEN 65W GaN Charger](https://www.ugreen.com/products/65w-gan-usb-c-fast-charger?variant=32127034654782)**

Since I have to go now, I will have to proceed testing later.

kind regards

Ralim, 2020-12-31:
@Airmonkdev 

PCB 99 means that your accelerometer was _not_ detected in boot.

Could you flash the older releases and check what PCB version it reports?

@discip 

Are you saying that the TS80P does not negotiate pd as reliably ? Are you seeing this because it's picking 9V or using the new debug menu option ?

I have a very similar charger here and will test tomorrow.

Airmonkdev, 2020-12-31:
Just flashed the v2.12 and PCB is 3

Thanks for support

Ralim, 2020-12-31:
> 
> 
> Just flashed the v2.12 and PCB is 3
> 
> Thanks for support

That would indicate that 2.12 also does not detect your accelerometer, so sleep modes were likely non functional before as well.

discip, 2020-12-31:
@Ralim
1. Yes indeed.
2.  I am using the debug menu.
And **`PWR`** is in most cases **1**.

Ralim, 2020-12-31:
@discip 
There has been no code changes between https://github.com/Ralim/IronOS/commit/b359e6e37e9c10a0e21c89eb67cd20184dbb7bd3 and the final release firmware?

If you set the unit to QC=9V, does it negotiate 9V or 12V?
How long does it take to negotiate the voltage?

Just trying to figure out if its showing the wrong `PWR` entry or if its actually using QC.

discip, 2020-12-31:
1. Strange? 🤷
2.1. If **QC** kicks in, it negotiates for **9V**.
If **PD => 12V**.
Judging by the debug menu.
2.2. That's really the riddle here, because I do not know, what's really going on in the background.
But what I can tell:
In **QC** it heats up significantly slower.
And: Sometimes it immediately, shows **12V**, sometimes it first shows **5V** and than climbs up to **9V** or **12V**.

Ralim, 2020-12-31:
Interesting indeed.

2.1 -> Makes sense
The pause at 5V then climb up is QC,as QC takes at least 1.2 seconds at a minimum.

Could you give this a try and check if you get any warnings about pd ic not detecting: https://github.com/Ralim/IronOS/actions/runs/455253208

discip, 2021-01-01:
@Ralim
First of all:
### Happy new year!
Sorry was late yesterday. 😊
This version does solve the **QC / PD** issue, however now the ***autostart-sleep*** function is 'broken'! 😵🤷‍♂️

regards

Ralim, 2021-01-02:
_autostart-sleep_ is very fragile to exactly how that particular accelerometer starts up and system noise. 

Have pushed a new commit to that branch, could you test if this is now good for you?

The temp jump may be the keep alive power pulse, could you test with that off if its on to see if thats the cause?

Airmonkdev, 2021-01-02:
At the moment I can't build the source code... can you please give me the EN hex file so I can flash it on my TS80P? :)

discip, 2021-01-02:
@Ralim 
Good morning sir,
thank you for being persistent! 👍

2. Unfortunately this commit did not solve the ***autostart-sleep** function*.
3. Seems to be true, since the previously described jumping behavior vanished.

I hope you do not get mad at me, for asking and 'reporting' (imho) continuously. 😅
I noticed one more thing, with **`RTip`** since the *jumping* is gone now:
The value drops over time.
For example: When I first calibrated the inserted tip **`RTip`** was somthing like **1500**, when I recalibrated the value decreased to **~1470** and by watching **`RTip`** in debug menu, one can observe, that the value is ever so slightly going down without being moved or even touched.

@Airmonkdev 
https://github.com/Ralim/IronOS/actions/runs/457430023

Ralim, 2021-01-02:
Ah bummer, auto start works for me here hmm...

Shall have to have another look 🤔

And yes it going down is actually normal! 🤯
Because the top is a thermocouple it's messing the temperature difference between the tip and the cold junction (inside handle).

As you use / hold the iron the handle warms up; which will therefore reduce the temperature delta, and therefore a smaller voltage is produced.

If you cooled the handle down(without cooling tip) it would go back up again.

discip, 2021-01-02:
1. Would appreciate that! 😁
2. Aha, that is weird, because I am not holding the Iron, it just sits on my desk, being stared at.
I am not monitoring the temperature in the room, but if it would go up, everything would get warmer (tip & handle) so *delta* should stay the same, right.
3. I am very glad you are seeing it that way! 👍

discip, 2021-01-02:
@Ralim
That's strange!
I just flashed another language of  [Add warnings for missing parts CI #118](https://github.com/Ralim/IronOS/actions/runs/457246634), and that worked with ***autostart-sleep***. 😵
However flashing the *EN* version, it than quits working.

Philipp-3, 2021-01-02:
Hello there!

first of all I want to thank Ralim for the nice project!
I have the same issue as Airmonkdev (PCB 99), and had it also with the previous versions. On original miniware firmware the accelerometer did work.
This made me looking if one of the supported accelerometer (MMA8652, LIS2DH12 or BMA223) is installed. See the picture of the top-side of the pub. Didn't find the IC, but perhaps it helps to you.
[image]

Ralim, 2021-01-02:
Hello there @Philipp-3 

Interesting, that does indeed look like a different IC juding by its top. (I havent seen a LIS with that etching to date).
They normally have a qr like code burned onto them. its the one to the left of the STM32.
(3219 is the USB-PD interface).

Gah thats frustrating if they have changed model again

Any chance you can possibly get a photo from the side that shows (roughly) how many pads on each edge?

Ill make an I2C scanner firmware build to try and figure out what its I2C address is at least.

discip, 2021-01-02:
My settings are:
Sleep temp|Sleep timeout|Shutdown timeout|Motion sensitivity
-|-|-|-
90°C|10S|7M|7

Ralim, 2021-01-02:
@discip Are you using start to sleep or start to off?

@Philipp-3 / @Airmonkdev 
Could you put the attached firmware on your unit, and power it up.
It should numbers for what ID's it finds.
When a number shows, can you note it down and then press any button and it should move on to a new number.
Keep going until it loops, hopefully you should find a total of 4 numbers. (two pairs of numbers next to each other).

If you could do this and report back the numbers it will help narrow the search space dramatically.
[TS80_I2C_Probe.zip](https://github.com/Ralim/IronOS/files/5760302/TS80_I2C_Probe.zip)

discip, 2021-01-02:
Sorry I missed to mention that: 😔
I am using the last setting **`O`** to make the iron sleep at room temperature.

discip, 2021-01-02:
@Ralim 
## 👍 Now it works with the ***EN*** version as well!

Ralim, 2021-01-02:
Ohhhh!!

Yes dragging straight out of the zip has caused me grief before, where I would get random flash failures.

Thinking about this, I have a suspicion that the firmware writes the blocks in the order they arrive.
Combined with hex not encoding the full address, getting an out of order block around a page change would corrupt flash.

If the zip software extracts directly without doing temp+move I could absolutely see that happening?

Philipp-3, 2021-01-02:
> @discip Are you using start to sleep or start to off?
>
> @Philipp-3 / @Airmonkdev
> Could you put the attached firmware on your unit, and power it up.
> It should numbers for what ID's it finds.
> When a [...]

Hello Ralim,
there are 4 numbers shown: 76, 77, 120 and 121.
If this doesn't help, I can try to figure out the IC's housing (unfortunately I just glued the display with double side sticky tape to the housing to avoid dust on the display :) )

leonardpitzu, 2021-01-02:
same numbers in my case - and obviously the same issue.
@Ralim - thanks for your support!

discip, 2021-01-02:
@Ralim
# Thank you very much for being patiently working this out! 👍❤

One thing that still remains though, is the fact, that the temperature still fluctuates between e.g.: **21°C** & **27°C**.
It does that not very frequently, but it does change just by sitting on the desk, without any intervention.
I will watch this behavior and try to find a pattern or cause by pure visual observation, since I am not skilled enough to do it on another level.

Ralim, 2021-01-02:
Alrighty, thank you everyone, waking up to all the useful information is fantastic ❤️ 

@discip thank you for the board image, absolutely makes it look like its a new spin they have done to down-cost.

Thank you @leonardpitzu  and @Philipp-3, those numbers indeed are extremely useful. Absolutely confirms they have changed the model without warning agian.

Going to move the accelerometer discussion over to #755 for clarity.

@discip Im not entirely sure of the exact cause of the changes with the power pulse turned off. Will require more investigation.

Airmonkdev, 2021-01-03:
Hey Ralim, sorry for the delay.
Same numbers also on mine: 76 77 10 121

Thanks man!

discip, 2021-01-04:
@Ralim
Are you planning to merge this branch into master, after you derived OPs issue into https://github.com/Ralim/IronOS/issues/755 and solved my irons "sleeping" issue as well?

Ralim, 2021-01-04:
Yes, I will probably merge the warnings branch in later tonight.
Going to merge it before the accelerometer support is done, but wont release until after #755  is done

discip, 2021-01-04:
By the way, as you mentioned accelerometer:
Is the later version of it better or worse? I did not get that! 😊

Ralim, 2021-01-04:
Generally they are all about the same.

It tends to be as a new company makes one for slightly cheaper miniware jumps to that model

Ralim, 2021-01-04:
Using the latest build from [here](https://github.com/Ralim/IronOS/actions/runs/460858679)
Is there anything else outstanding that people know about for v2.13?

discip, 2021-01-04:
You are fast! 😲
I just flashed https://github.com/Ralim/IronOS/commit/c358083cc137c74740be4dd43c3552c1a1e66e35.

Let me try out the latest as well.

Ralim, 2021-01-04:
Trying to get the outstanding bugs list smaller before I get busy with work again 😂

Airmonkdev, 2021-01-04:
Thank you so much for all you've done!

Just opened a new issue for the Apple reboot and crossed my fingers. :)

discip, 2021-01-06:
@Ralim
Good afternoon,
as I had to find out the hard way, that **zipped files have to be unpacked before dragging them *'into the iron'***:
>I think the problem was, that I dragged the EN version direct from the .zip

and you confirmed that:
> Yes dragging straight out of the zip has caused me grief before, where I would get random flash failures.

and @Urus74 confirmed that as well:
https://github.com/Ralim/IronOS/issues/639#issuecomment-754905255

I think it would be nice, if this would be added to the documentation somewhere, I am not aware of such an entry, already being placed there.

Vinigas, 2021-01-13:
TS80P straight outta official box, v2.13 didn't detected accelerometer.

After building and flashing current master ( v2.13.2052f30 ), everything working. Well, except maybe for custom-boot logo, some pixels looks like swapped in the matrix.
**PCB 4**

discip, 2021-01-16:
@Ralim
Sorry I have to bother you *again*, but I am still not able to wrap my head around the temperature drift.
I know, you have explained, that before https://github.com/Ralim/IronOS/issues/749#issuecomment-753453283:
>And yes it going down is actually normal! 🤯
Because the top is a thermocouple it's messing the temperature difference between the tip and the cold junction (inside handle).
>
>As you use / hold the iron the handle warms up; which will therefore reduce the temperature delta, and therefore a smaller voltage is produced.
>
>If you cooled the handle down(without cooling tip) it would go back up again.

... but how am I able to tell, whether the iron is showing the correct temperature?
Because in my case the *'drift'* amounts to almost **10°C**.

Is there no other way to determine the tip temperature, without having to consider the cold junction in the handle?

I am apologizing in advance, for asking you to answer questions like that. 😌

thanks in advance

<details>
  <summary>
    <b>setup</b>
  </summary>

iron  |  Firmware Version  |  PCB Version  |  Power Supply
-|-|-|-
**TS80P**|**https://github.com/Ralim/IronOS/commit/6d56ba3322959afdac235f8c955cfc69564f2a9c**  |**2**|**[UGREEN 65W GaN Charger](https://www.ugreen.com/products/65w-gan-usb-c-fast-charger?variant=32127034654782)**

</details>

Ralim, 2021-01-16:
The only way that is 100% reliable is to use an external tip temperature checker.

Rtip is a raw value, so you should not be concerned by it drifting around of the ctip is holding steady.

discip, 2021-01-16:
Good evening,
yes that is exactly what I am concerned about.
Because I am not only seeing **`RTip`** *changing* , but it is the temperature, which is shown on the detailed idle screen (*Seems to be the same as **`CTip`**.*), that is ***jumping*** from **27°C** to **34°C**.

Here is what I did to observe that behavior:
1. Provide the iron with power **(Not heating the tip.)**.
2. Calibrate the tip.
3. Go to idle (detailed) screen, or in ***debug menu*** **`CTip`** and note the temperature.
4. Surround the iron, without touching the tip, with your palm.
5. Watch the temperature climb from **26°C** to **27°C**, **28°C** and suddenly to **34°C**! 🤷‍♂️
6. Let go of the iron and put it on your desk to let the handle *'cool down'* to room temperature again.
7. Watch the temperature descend from **34°C** to **33°C**  and suddenly to **26°C**! 🤷‍♂️

Hopefully this helps, to get an idea, what I am referring to.

Thanks in advance
regards

discip, 2021-01-17:
@Ralim
I hope you are not upset by now. 😅

There is just one more thing I have observed, which I would like to share (Most likely you know this already, because it is by design, maybe.):
After having calibrated the tip, as you know **`RTip`** keeps descending, *in my opinion*, to infinity.
(Was not patient enough, to wait, until it stops descending.)

I noticed, that the temperature only jumps when the calibrated **`RTip`** is reached:
1. Let the iron sit and wait for **`RTip`** to go down for some time,
2. now calibrate the tip,
3. unplug and re-plug the iron,
4. it shows the ***actually real*** ***room temperature*** (Confirmed with *external thermometer*.),
until **`RTip`**, the iron was previously calibrated at, is reached.
5. Now the **JUMP** occurs (eg: **22°C** => **30°C**).

Is this by design, meaning the behavior described here: [Temperature.md](https://github.com/Ralim/IronOS/blob/master/Documentation/Temperature.md), or is this a jump in the lookup table, or anything else?

### thank you very much in advance
kind regards

Ralim, 2021-06-25:
Apologies I completely missed responding to this.

The jump you are observing _shouldn't_ really occur, and _might_ be a bug but im not 1000% certain of the cause off the top of my head.

Definitely there will be some jump as we mask to 0 offset while RTip is < calibrated offset.

discip, 2021-06-25:
@Ralim
Good evening Sir,
you have absolutely no reason to apologize. But at the same time, I do not refuse or depreciate the good manner you are proving to have. I think, I would have reacted similarly if not the same way. 😃👍

1. Do you see any chance to look into that?

2. Am I the only one seeing this behavior?

3. Additionally to the above mentioned, I am also seeing instabilities when heated up to the desired temperature. But since I do not know which range I have to expect, I do not know, if it is ok.

e.g.:
Heating up and letting the iron sit there without touching neither solder, nor any conductive surface.
Set temp is 330°C.
The temperature jumps between **`33X°C`** and **`35X°C`**(Since I could not figure out the last digit, I replaced it with 'X'.)

thanks in advance
kind regards
<details>
  <summary>
    <b>setup</b>
  </summary>

iron  |  Firmware Version  |  PCB Version  |  Power Supply
-|-|-|-
**TS80P**|**https://github.com/Ralim/IronOS/commit/9ed1c1fc043736af873eb8164af1b62dfc7cb4c5**  |**2**|**[UGREEN 65W GaN Charger](https://www.ugreen.com/products/65w-gan-usb-c-fast-charger?variant=32127034654782)**

</details>

Ralim, 2021-06-27:
@discip 

> 
>     1. Do you see any chance to look into that?
> 
>     2. Am I the only one seeing this behavior?

The jumpiness has been noted elsewhere, you are not alone in spotting the way the temp sensor reacts.
Seeing the jump around the cal point _shouldn't_ occur, but I'm currently thinking its probably exposing a weakness of the current cal routine, in that we are lumping a few sources of error into one, when they probably shouldnt be.
I need to figure out a good way to thermally cycle an iron to get a feel for these 😂 

I don't have a timeline yet as I'm not sure the best way to test and model the issue, what I'm most suspect of is that there is an increasing error that is a function of handle temperature that is not calculated _yet_ that really should be. Though I'm also unclear of a good way to derive said offset without just empirically measuring it.

> 
>     3. Additionally to the above mentioned, I am also seeing instabilities when heated up to the desired temperature. But since I do not know which range I have to expect, I do not know, if it is ok.

It's _ok_ but I do need to add in some code to not visually show these. Mostly its just the side effect of having the same piece of wire be your temp sensor and your heating element.

> 
> 
> e.g.:
> Heating up and letting the iron sit there without touching neither solder, nor any conductive surface.
> Set temp is 330°C.
> The temperature jumps between **`33X°C`** and **`35X°C`**(Since I could not figure out the last digit, I replaced it with 'X'.)
> 
> thanks in advance
> kind regards
> setup

discip, 2022-02-12:
@Airmonkdev
Please close this issue, if this is solved. 😊

thank you

closed 2022-02-15 by Ralim