Ralim/IronOS #751: TS100 3.45 logo issue

lestatsmods, 2020-12-31:
the .exe custom logo creator makes a .hex and when u load it to the iron it dose say .rdy...
however the logo wont show and it seems to move all the setting on the screen to the far right cutting off all the wording, so i have to reflask the EN.hex to fix this... Drib 2 youtube vid provides his logos.. and BOOM they work... but mine.. nope....

it got to the point my iron WOULD NOT POWER at all but will be reconised via the pc... but the EN.hex always failed in the transfer.... i then ordered a 2nd thats on it way now.... this eve i thought.. ill try it again... and the EN.hex file transfered and works again!!!!! but still no logo working.... (yes i have a second on it way still)

i think i read somewhere there was an update to the .exe progam but ill be damned if i can find the updated version

  - Firmware Version: 3.45
  - PCB Version: (1/2) ???
  - Power Supply (Voltage and Current Rating) : 12V

Ralim, 2020-12-31:
There is no exe version anymore, the only tool that is currently supported is the python conversion script.

lestatsmods, 2020-12-31:
1. thank you.
2. no to dive into more stuff i have no clue about ahahah. wish me luck

lestatsmods, 2020-12-31:
ok i have NO clue about coding AT ALL,...... would you mind making this a hex file....... pretty please 

[image]

Ralim, 2020-12-31:
[logo.zip](https://github.com/Ralim/IronOS/files/5758435/logo.zip)

lestatsmods, 2020-12-31:
omg THANK YOU!!!!!! ended my year! thank you once again

closed 2020-12-31 by Ralim