Ralim/IronOS #753: data logging support

dhiltonp, 2021-01-01:
I know we can'tn do usb logging with the ts100 because the data lines don't go to the MCU. 

Will it be possible to log data out of the usb-c port on the pinecil? 

I'm thinking of ways to improve data collection for temp control validation, maybe making a test fixture with an Arduino logging temps and amps. Seeing exactly the same temp data as we're processing on the MCU would be really handy... 

Maybe we can write values to some spare pins if nothing else.

Ralim, 2021-01-01:
Ts100 does have them connected but nowhere near enough flash space for a USB stack.

The Pinecil has a bunch of gpio broken out on the usb-C _and_ the USB port. So definitely can on the Pinecil. (And it has double the flash + no bootloader)

If you want to do a bunch of tuning I can possibly send you one?

dhiltonp, 2021-01-02:
That would be great, yeah!

I could set up something to simultaneously record the internal thermocouple readings as well as read from an external thermocouple which would really help.

For general use, logging over USB-C seems most convenient, but in order to test at a variety of voltages it seems that it would be easiest to provide power via barrel jack and read from the breakout board. What do you think?

Ralim, 2021-01-02:
My inital plan is to add support for sending state information out the uart as the unit is operating.
Mostly as I dont really have time to setup the usb stack on the short term.

Other thought is that very few devices can output 20V over USB-C while also having the usb data lines connected. :/

My thought is uart on breakout and PD or DC in for power.

dhiltonp, 2021-01-02:
Sounds good!

OndroNR, 2021-01-16:
I am thinking of automated fume extractor fan triggering. This would require special "base station" with USB-C connector and some very careful high voltage power handling, so it would not fry random connected devices. I am even considering USB host counterpart, so fully connected cables would not be required.