Ralim/IronOS #755: Support new Accelerometer in newer TS80P models

Ralim, 2021-01-02:
This is a breakout from #749 
As the issue is unreleated to the latest release.

It looks like there is a new model of accelerometer using I2C address 0x3C.
Looks like this _could be_ `MSA301`

Plan:

* [ ] Make firmware to readout something to confirm its this model
* [ ] Make test firmware for reading out the 3 axis of information
* [ ] Make test firmware to read out orientation values
* [ ] Setup driver class to decode ^ into expected formats
* [ ] Test works as expected
* [ ] Make a new release

Ralim, 2021-01-03:
[TS80_read_id.zip](https://github.com/Ralim/IronOS/files/5761025/TS80_read_id.zip)

Here is the first step, this will read register `0x01` which is the id reg. and should return `0x13` (in decimal this is 19).

Could someout out of ( @leonardpitzu, @Philipp-3 and @Airmonkdev) please flash this and test it and show us what number shows up ?

leonardpitzu, 2021-01-03:
it reads `19`

Philipp-3, 2021-01-03:
Can confirm leonrdpitzu reading at mine

Ralim, 2021-01-03:
Fantastic.

Will write up a rough driver for this when U get a chance. 

Will be back with a tester once that's done

Ralim, 2021-01-03:
Alrighty, attempt one (possibly buggy).

This should show 4 numbers on the screen.

1. Are they non-zero
2. Do the change with orientation
3. If they do change 🎉 , can you record approx values for their magnitude in the laying flat and in soldering position
4. The last one (Orientation) is a bit fiddly, but, its used for screen orientation rotation support.
-> Could you rotate the unit around and record the numbers you get for iron in left vs right hand, and what the transition is.

For example, it could show 3 for left hand, 2 for flat (in the middle) and 1 for right handed for example.

[TS80_Read_MSA301_regs.zip](https://github.com/Ralim/IronOS/files/5761480/TS80_Read_MSA301_regs.zip)

The numbers are laid out as:

```
Accel X        Accel Y
Accel Z        Orientation
```

leonardpitzu, 2021-01-03:
1. nope - it's just 0's all the way
2. nope

Philipp-3, 2021-01-03:
Same on mine, independent from orientation:
000000  000000
000000  000

Airmonkdev, 2021-01-03:
Hey Ralim,
just tried the read_id and it return 19 (0x13).

All zero, as Philipp-3 for the regs

Ralim, 2021-01-04:
Ah bummer, probably my mistake. Will try and fix shortly

Ralim, 2021-01-04:
Attempt 2, less typo's this time :)
[MSA301_attempt_2.zip](https://github.com/Ralim/IronOS/files/5763458/MSA301_attempt_2.zip)

leonardpitzu, 2021-01-04:
1. yes (gr8 work mate! i envy you right at this moment - used to do embedded stuff and now i have 0 time to do so)
2. yes

flat on the table (tip pointing right - last 2 digits jumping around):
```
065319 0653334
009780 000
```

right hand soldering:
```
062760 005386
007710 000
```

left hand soldering:
```
002614 004612
007950 000
```

thus orientation stays `000`

Ralim, 2021-01-04:
[TS80_attempt3.zip](https://github.com/Ralim/IronOS/files/5763609/TS80_attempt3.zip)

Fantastic,
Could you test this version to see if the orientation changes now?

Airmonkdev, 2021-01-04:
Tried with attempt 2 and it works!
Not simple to get the number but here there is my output:

flat on the table tip pointing left:
0653XX 0653XX
0084XX 000

please consider that when flat in the table it is nor real flat because of the holder shape (where you insert the tip the diameter is bigger than the end of the iron where sit the USB-C port).
I've tried to "level" the solder and it seems to me that when really flat the second number is going to 0:
0653XX 000000
0084XX 000

...the first number changes when I rotate the solder (immagine to use it like a screwdriver) and it seems to going to 0 when flat, increase from 0 to 065XXX when turning clockwise and decrease from 065XXX to 0 when turning counterclockwise
Also the third number changes when turning the iron but not sure how. The last one is always 000

Airmonkdev, 2021-01-04:
Just flashed the attemp3 and yes! Now also the 4th number changes: 
solder vertically with the tip pointing up:  080 
solder vertically with the tip at 90degree right: 112
solder vertically with the tip at 90degree left: 96
solder vertically with the tip pointing down: 064

PS: sorry to put this comment here but I can't use the TS80P connecting it to the USB-C port (thunderbolt 3) of my MacBook Pro 13" 2020: I can flash it but when I connect it in normal mode it seems to reboot...

Philipp-3, 2021-01-04:
Great work!

seems to be 16bit signed  numbers.
- tip is y-direction with positive values the facing downwards, 1g around 8000
- display is z-direction with positive values when facing upwards, 1g around 7200
- x-direction according to righthand-rule, 1g around 8200

Philipp-3, 2021-01-04:
- left hand soldering: 112
- right hand soldering: 096
- display straight upwards, soldering: 064
- display straight upwards, tip lifted uo like in simple stand: 080

leonardpitzu, 2021-01-04:
flat on the table - tip left: 096
flat on the table - tip right: 112
right hand soldering (~45 deg in right hand): 064
left hand soldering (~45 deg in right hand): 064
tip upwards: 080
[image]
[image]

Ralim, 2021-01-04:
Fantastic ❤️ 

Alrighty, this attached firmware _should_ act basically the same.
If you could flash it and thumbs up if it acts about the same would be great. (with exception of orientation below)
(This is validating that the bulk register read works, and is last step before testing on real firmware).

The orientation numbers will be different, and instead should be showing as:
- left handed 0
- right handed 1
- all others 2 (these are ignored later on)

[TS80_MS301_Validate.zip](https://github.com/Ralim/IronOS/files/5764031/TS80_MS301_Validate.zip)

@Airmonkdev 
PS: sorry to put this comment here but I can't use the TS80P connecting it to the USB-C port (thunderbolt 3) of my MacBook Pro 13" 2020: I can flash it but when I connect it in normal mode it seems to reboot...

Apple stuff is a more fussy than most things, and have not been able to capture this behaviour on my datalogger so far.
😢

leonardpitzu, 2021-01-04:
with the iron flat on the table, I get:
- tip pointing left (right hand soldering): 1
- tip pointing right (left hand soldering): 0
- everything else: 3

with the iron in the soldering position (hand resting on the table and holding the iron like a "pen"):
- tip pointing left (right hand soldering): 3
- tip pointing right (left hand soldering): 3
- everything else: 3

Airmonkdev, 2021-01-04:
Just flashed the Validate:
it seems to works fine changing from 0 to 1 when on left or right hand but sometimes it show 3.
It seems that the 3 value is shown when I turn little bit the iron while on my hand.

PS: may I open an issue for the Apple stuff? Do you think I can help you to solve also this issue with some tests or debug version?

Ralim, 2021-01-04:
Could you please test this and use the debug menu to check that the version is detected as 4 (not 99)
If that works, could you look at the mov time in the debug menu and it should update the number when the unit moves, and when it sits still it should stay at the same number.

if that works, could you try putting the screen rotation to auto and flipping the unit around and seeing how the screen behaves.

If this works as expected then support should be done (i hope....)

[TS80P_EN.zip](https://github.com/Ralim/IronOS/files/5764216/TS80P_EN.zip)

>PS: may I open an issue for the Apple stuff? Do you think I can help you to solve also this issue with some tests or debug version?

You can, but I cant promise much here because I dont really have access to any apple devices to test against.

leonardpitzu, 2021-01-04:
1. PCB is 40
2. move time is correctly updating (incrementing)
3. screen rotation works

thanks a bunch!

closed 2021-01-04 by Ralim

Airmonkdev, 2021-01-04:
Yeees!!! Everything seems to work fine now!

Really thanks for the great work!

Philipp-3, 2021-01-04:
Just tested on mine, works great! Such a pleasure to use the soldering iron now.
Many thanks for the fantastic service, Ralim!

PR #761 (2021-01-04): Feat add msa301
Adding support for the new MSA301 for the newer 1.2 version of the TS80P

Closes #755