Ralim/IronOS #757: Hakko tips having slow regulation response, drops 20 C before heating up again

robinhellmers, 2021-01-03:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Question]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

With Hakko T15 tips, the temperature slowly drops by 20 C and then does a fast heat up with 10 C overshoot by what the display shoes. Do anyone have the same problem?

* **What is the expected behavior?**

Faster regulation of the temperature, not having to drop by 20 C before heating it up again.

***Steps to reproduce the bug:***

- Use Hakko T15 tip.
- Do calibration.
- Set temperature and let it heat up.
- Now the temperature will start to drop.

* **What is the motivation / use case for changing the behavior?**

I want a somewhat stable temperature, even if the actual temperature offset is something else than what is measured by the iron.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.13
  - PCB Version: (2)
  - Power Supply (Voltage and Current Rating) : Benchtop power supply 32V, 5A

Ralim, 2021-01-04:
Hello,
I have a pair of hakko T15 tips here that I use, and I have not observed this issue.
Are you sure you have good contact inside the ts100 on the tips contacts?
Is this issue present with ts100 tips?

What temperature are you soldering at?

robinhellmers, 2021-01-04:
I just opened it up, made sure that the contact points were at the right place and I still have an issue. Seems to just be an issue with the Hakko tips and not the original TS100 one i got. With the original one, it just overshoots like 10 C then comes down and settles at the set temperature. 

This time I tested with the Hakko tips, I got a bit different result. Before heating, I can see that the TS100 firmware reads 22-23 C which is room temperature. Then when entering soldering mode, the read temperature jumps down to 0 C then starts to rise from there. Then it goes like 20-30 C higher than the set temperature and then slowly drops like 20 C and then heat up to a level of about 20-30 C above the set temperature.

I have tried to set the temperature in the range of about 280-320 C.

Ralim, 2021-01-04:
Hi,

Can you confirm this still exists after resetting to default settings and _not_ doing a temperature calibration.
A bad temp calibration could cause this

robinhellmers, 2021-01-05:
Hi,

I can confirm that this problem still exists after doing a factory reset. Here is also a video showcasing my problem, I just activated extra information on the idle screen. Notice the drop to almost 0 C before rising.

https://user-images.githubusercontent.com/38323910/103622989-c01f3b80-4f37-11eb-9b1f-7b4eedf11ab7.mov

alvinhochun, 2021-01-05:
I think I might have seen a similar thing on a Pinecil with some of the tips (some came with the Pinecil, some are third-party TS100-compatible tips). This was right after flashing the 2.13 firmware. I don't recall seeing it with the factory firmware when I did some soldering.

I set the temperature to 80°C since I only wanted to test that the tips that I just got heat up but not too hot. When I tested one of the tips, the temperature dropped for a bit (think about 10°C) as soon as I started heating, then it overshot by 10-20°C. Then I observed the same behaviour with one of the tips that came with the Pinecil. Not all the tips showed this behaviour.

I can do a bit more real testing in a couple of days, but I don't have a soldering iron thermocouple so I won't be able to give much info.

robinhellmers, 2021-01-05:
> I think I might have seen a similar thing on a Pinecil with some of the tips (some came with the Pinecil, some are third-party TS100-compatible tips). This was right after flashing the 2.13 [...]

Thank you for your input. I don't have a thermocouple either, so I cannot verify what the actual temperature is. The main problem is that it won't stabilize though. I will test to flash an older firmware.

Ralim, 2021-01-05:
What is your ambient room temperature?

Could you provide all of the clashes in the debugging menu when you have a chance?

This feels like something not working in the cold junction compensation.

robinhellmers, 2021-01-06:
I don't have a thermometer, but about 20-24 C room temperature.

Here's **V2.13**, no heat up before. Neither calibration of temperature or voltage:
- v2.13.28736CB, 31-13-20
- HW G 171
- HW M 190
- HW P 96
- Move 436
- RTip 830
- CTip 28
- CHan 290
- Vin 23.8 (But it actually is 24.0 V)
- PCB 2
- PWR 0
- Max 410

Heated it up with 320 C set. Neither calibration of temperature or voltage:
- HW G 171
- HW M 190
- HW P 94
- Move 1316
- RTip ~7300+
- CTip ~307
- CHan 332
- Vin 23.7 (But it actually is 24.0 V)
- Max 414

Reheated the same tip, from some temperature. But calibrated voltage to 24.0 V, no temperature calibration:
- HW G 171
- HW M 190
- HW P 94
- Move 2252
- RTip ~7390+
- CTip ~311
- CHan 346
- Vin 24.0 (But it actually is 24.0 V)
- Max 415

New tip, not heated. Calibrated both voltage and temperature:
- HW G 153
- HW M 190
- HW P 94
- Move 3344
- RTip ~805
- CTip ~31
- CHan 314
- Vin 24.0 (But it actually is 24.0 V)
- Max 414

Same as before, but heated up:
- HW G 153
- HW M 190
- HW P 94
- Move 3762
- RTip ~7200
- CTip ~307
- CHan 345
- Vin 24.0 (But it actually is 24.0 V)
- Max 417

Ralim, 2021-01-06:
Aha thank you, that is very helpful.

What is most likely occuring is that the default calibrated uV offset for the tip calibration is not correct for your unit/tips.
Depending on exactly how the tips are constructed this can vary.

If you let the iron cool down so tip temp == handle temp (ie leave unplugged for an hour)
And then run the tip calibration, what number is shown at the end ?

discip, 2021-01-17:
@robinhellmers
Are you not interested in getting help anymore? 
Please provide the data, @Ralim asked for, if you want that issue to be solved. 😊
thanks

robinhellmers, 2021-01-18:
@Ralim @discip Sorry for not responding. I am a bit short on time as I am in my final part of my master thesis. I will absolutely provide the data as soon as possible.

discip, 2021-01-20:
@robinhellmers
This was not intended to make you panic, but if there is no response at all, people start thinking, that this issue might not be that important.
You set the right priorities. 👍
kind regards

thrawn86, 2021-02-08:
I believe I have the same issue, but with an original ts100 bc2 tip.  Cal value from cold gives 933.

[https://www.dropbox.com/preview/Camera%20Uploads/ts100%20tip%20temp%20swings.mp4?dl=1](url)

Ralim, 2022-02-15:
Would love if you could retest a newer firmware (2.17 onwards) to see if this is still present:)

discip, 2022-04-23:
@robinhellmers
Are you still facing this issue, after updating to the latest release?
If not, please don't forget to close this issue. 😊

thanks

thrawn86, 2022-04-26:
I also still have the same issue I posted about before on the latest firmware.  Sorry I missed the request for an update before.

https://user-images.githubusercontent.com/48975359/165208574-9f5b5acd-d413-4208-84a2-f7bd2206f979.mp4

robinhellmers, 2023-05-04:
@thrawn86 Have you any updates on your end? As I would like to use the Hakko tips I have, I am considering buying the Pinecil V2 and try it out.

thrawn86, 2023-05-09:
> @thrawn86 Have you any updates on your end? As I would like to use the Hakko tips I have, I am considering buying the Pinecil V2 and try it out.

same issue here on newer fw, but i bought some new tips that do work fine