Ralim/IronOS #765: cool down blink flashes really fast

Dag0d, 2021-01-04:
This is a Feature Request

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
Since the last release the cool down blink flashes really fast, way faster than before.
With 2.12 it was WAY slower IMO even too slow.
I think something in the middle would be the best

* **What is the expected behavior?**
Blinking slower

***Steps to reproduce the bug:***
enable cool down blinking

* **What is the motivation / use case for changing the behavior?**
Better readability if the blinking is a bit slower

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.13 and I also tried a freshly compiled firmware
  - PCB Version: 2 (TS100)
  - Power Supply (Voltage and Current Rating) : 12v DC and 20v DC

Firebie, 2021-01-04:
```diff
-					if (!(systemSettings.coolingTempBlink && (xTaskGetTickCount() % 250 < 160)))
+					if (!(systemSettings.coolingTempBlink && (xTaskGetTickCount() % 270 < 160)))
```

discip, 2021-01-04:
@Firebie
Are you planning to create an PR?

Dag0d, 2021-01-04:
Now as I know where to find that setting I can take care of the PR @discip

discip, 2021-01-04:
Nice!

Ralim, 2021-01-04:
Thanks guys for getting the PR ready overnight 🤟🏼

closed 2021-01-04 by Ralim