Ralim/IronOS #768: v2.13 Auto Display Orientation issue ?!

kokein, 2021-01-04:
_This is a [Bug]_

* **I have**
  - [X] Searched previous issues 
  - [X] This is in _this_ firmware, not vendor "offical" firmware
  - [X] This is not a bug in the vendor bootloader (aka DFU)
  - [X] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
*  v2.13 Auto Display Orientation issue was ok but now is is oposite 

* **What is the expected behavior?**
* It shoulud change display other way

***Steps to reproduce the bug:***
- Turn on the Auto Display Orientation and check it yourself :) 

* **What is the motivation / use case for changing the behavior?**
* 

* **What are you running:**
* TS80, EN ,  v2.13 

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number v2.13.28736CB 31.12.2020 

  - Firmware Version: 2.13
  - PCB Version: (1/2) 2
  - Power Supply (Voltage and Current Rating) : USB C

discip, 2021-01-04:
Confirmed!
iron|Firmware Version|PCB Version|Power Supply
-|-|-|-
**TS80P**|**https://github.com/Ralim/IronOS/commit/0b623544663ebfab0f6dcf49529720578eed532e**|**2**|**[UGREEN 65W GaN Charger](https://www.ugreen.com/products/65w-gan-usb-c-fast-charger?variant=32127034654782)**

discip, 2021-01-07:
@Ralim 
I do not know for sure, but to me it seems, as if **`L`** & **`R`** are swapped in **`'auto mode'`** somehow.

commit b961311a (2021-01-16): Fix ifdef for inverted accel in TS80/TS80P

This should fix #768
closed 2021-01-16 by Ralim