Ralim/IronOS #772: Standby mode when reaching temperature

BytheWWW, 2021-01-06:
I have successfully updated the firmware, but when the soldering iron reaches the desired temperature it goes into standby mode (and thus cools down again) 

This happens every time I try to go into solder mode. Does anyone else have this problem?

Ralim, 2021-01-06:
Hello,

What model soldering iron do you have?

What version is detected in the debug menu?

Have you changed the motion sensitivity?

BytheWWW, 2021-01-06:
@Ralim Thanks for your quick response. The model is SQ-001 and the version is V2.13.28736CB. I didn't change the sensitivity (7)

Ralim, 2021-01-06:
Thanks.

If you scroll through the debug menu, there is a model number that is detected for your unit, it should be 1/2/3/4/99 ?

BytheWWW, 2021-01-06:
How do I enter the debug mode? (Sorry, bit lazy to look it up)

Ralim, 2021-01-06:
https://github.com/Ralim/IronOS/blob/master/Documentation/DebugMenu.md

BytheWWW, 2021-01-06:
It says 1

Ralim, 2021-01-06:
Excellent,

So in theory the movement detection is running.

If you look in that many at the mov timer, it will update to a higher number when there is movement.

If this does not update when you move the iron as if you were soldering you may need to increase the sensitivity in the menu.

That said, if your powering the iron from a 18-24V supply it should only take 5-10 seconds to warm up so it shouldn't be timing out on you.

BytheWWW, 2021-01-06:
It does register movement and I have a sufficient power supply (19v, 6.3A) It warm up quickly but then goes to standby once it reaches the set temperature. I must also mention the A button has to be clicked very shortly to get it into soldering mode. I can make a video if you want.

Ralim, 2021-01-06:
A video could be helpful.

You could also try resetting settings to defaults to make sure it's not something that's set to the wrong thing.

BytheWWW, 2021-01-06:
Did a factory reset but that didn't help unfortunately. Here's the video. As you can see it's pretty hard to get it into solder mode which might be the problem, although when it does get into solder mode it should stay there...

https://user-images.githubusercontent.com/7500149/103764415-b7079a80-501b-11eb-99cc-692b6516f78b.mp4

paulfertser, 2021-01-06:
Hey,

On Wed, Jan 06, 2021 at 03:37:32AM -0800, Koen Schuit wrote:
> [1]https://user-images.githubusercontent.com/7500149/103764415-b7079a80-501b-11eb-99cc-692b6516f78b.mp4

Kinda looks like rebooting. Have you tried the same after limiting
power to e.g. 15 W?

BytheWWW, 2021-01-06:
Sorry, can't seem to find how to change the wattage 🤦

paulfertser, 2021-01-06:
On Wed, Jan 06, 2021 at 03:55:12AM -0800, Koen Schuit wrote:
> Sorry, can't seem to find how to change the wattage 🤦

Advanced settings -> power limit

BytheWWW, 2021-01-06:
Did some tests but the power limit didn't matter. So, to sum it up:

Power supply 19V 6.3A doesn't work (strangely enough)
Power supply 12V 2A -> max power: 16.5W, reaches 415°C @ approx. 8W
Power supply 14V 4.14A -> max power 23.1W, reaches 418°C @ approx. 9W

howie-s, 2021-03-01:
Has this ever been figured out? I too have a SQ-001 Iron, and as soon as I enter solder mode it heats up to the setting temp then reboots. This behavior is consistent. Tried various settings as mentioned previously such as power limiting, increasing motion sensitivity, but same thing.

Iron Specs
--------------------------------------------
-  PCB = 1 (MMA8652)
-  Firmware = v2.14.2425902, 17-01-21
-  Motion Sensitivity = 7 (Default)
-  Vin = 19.4v
--------------------------------------------
Any thoughts or new info on this issue?

https://user-images.githubusercontent.com/78635001/109487351-18fbe380-7a52-11eb-9343-3cfd9e45dfa1.mp4

--------------------------------------------

@BytheWWW Your power tests on that iron are gonna be limited to 65W, so 19V @ 6.3A would be ~120W which is well beyond the capability of that iron. For example, the DC power supply I am using to power the SQ-001 is a 19V @ 3.42A supply (The ratings printed on it), which gives me the full power capability of the iron (19V)x(3.42A) = ~65W of power.

BytheWWW, 2021-03-01:
@howie-s I'm using a 58W adapter (14V at 4.134A) at this moment which works fine. Will have a look in the thrift store once they open again to see if I can find one closer to 65W at 19V.

howie-s, 2021-03-01:
@BytheWWW 
As long as you are getting c,lose to 65W it really don't matter what voltage your supply is, provided your supply is within the operating range of the iron (12V - 24V). Your supply is (14V)x(4.134A) = 58W, that is plenty! I wouldn't buy another supply for such a small gain unless you really needed to! I acquired the supply and iron together as a set so the supply was tightly matched with the iron. Save your money and use it to build something awesome that will make you happy!

Anyway, I am kind of frustrated that the iron keeps rebooting after heating up, which seems closely related to your issue. I would like to devote some time real soon to figure out something. I hate the stock firmware for this iron. I've thought about sequentially downgrading the firmware to see if the issue has always existed with this firmware for this particular iron.

BytheWWW, 2021-03-01:
@howie-s My problem went away when I used a different power supply so I don't think i can't help you any further with this issue. You could try a different power supply, just in case....?

howie-s, 2021-03-02:
@BytheWWW 
I didn't ask for your help, although the information that changing the power supply is a good start before "blaming" the firmware per se, although my issue may be closely related, it is not EXACT to the issue you were experiencing. I don't have another suitable DC source on hand to try, so I will prep a battery pack and use that. Happy soldering!

Ralim, 2021-03-03:
This would sound like either a power supply that isnt handling it will (batteries are great to try as they tend to be low noise).
For the system to reset like that its either:
* watchdog timed out
* Power supply was unstable

Watchdog is controlled by the PID loop that regulates the heater; its deliberately designed to not reply on calls that can block so it _shouldnt_ stall out and cause a WDT reset. Ergo why power supply is the most likely cause of these issues.
If you are still experiencing this issue on batteries it is also worth trying different temp set points.

That also said, the power supplies that were "paired" with a lot of the clone Irons that I have seen have _not_ handled the high repetition rate of the load from the firmware well.

howie-s, 2021-03-03:
@BytheWWW was right on the money about using a different supply, although he failed to share that he solved his problem by switching the supply before I mentioned it, leading me to believe the problem still existed for him as well.

@Ralim, At first I was thinking that maybe something became broken along the way with how this clone uses the firmware as following the progression of this thread it appeared that different supplies were likely being utilized by everyone experiencing this problem.

I constructed a down and dirty battery pack and operation was flawless (no surprise here). After tearing into the supply and getting it on the scope I did some testing at different settings and scenarios, and indeed it cannot handle the demands of the iron. This particular supply is your run-of-the-mill super cheap smps utilizing a simple oscillator block to drive a single transistor to transformer before being filtered and out to load. After the iron heats up to near the set temperature, the supply's feedback reference becomes unstable and causes duty cycle correction failure, ultimately leading to the iron resetting.

I absolutely don't like the stock firmware of this iron and so glad I found your firmware to make the experience a joy. Very nice work, which is appreciated greatly.

Ralim, 2021-03-03:
Thank you for getting back to us about the cause 😊

It's hard to point to supplies easily as it doesn't _look like_ it at first; but time and time again it turns out to be a supply going unstable.

Would love a photo / any measurements you made of the supply to satisfy my curiosity (No pressure though!).

Glad to hear you like the firmware.

2m, 2021-07-10:
I also started getting reboots after I flashed to `v2.15`.

I was not getting reboots (or at least I did not notice) with the original firmware. I am using PD65W port of [PinePower](https://www.pine64.org/pinepowerdesktop/) as the power source.

Tried limiting power to 15W, but still getting reboots when the tip reaches soldering temperature (350C).

Ralim, 2021-07-10:
@2m 
To confirm, which model of soldering iron are you using?
I find this interesting, as my _main_ power supply I use for testing is the PinePower.
It is however notable that the PinePower is **not** grounded. I have found that when its floating on the hot side, touching the iron to some things will grounded it and cause emi that causes resets in the pd controller.

2m, 2021-07-10:
Thanks for getting back to me so fast. I am using Pinecil. There is a separate ground connector on the soldering iron itself. But I have not connected it to anything.

Ralim, 2021-07-14:
@2m 

You shouldn't need the extra ground really, but I would suggest testing using a different power supply, as this may be the cause. You could also test flipping the figure-8 cable on the back to swap hot/cold lines and see if that makes a difference.

2m, 2021-07-15:
Tried flipping the figure-8 connector and that did not produce different results.

With other power supplies (HyperJuice GaN 100W USB-C Charger, and charger for Google Pixel 5) Pinecil works without problems.

Ralim, 2021-07-23:
@2m If its only reproducable with the one power supply it may just be that power supply sadly :(

Ralim, 2021-10-10:
> > It is however notable that the PinePower is **not** grounded. I have found that when its floating on the hot side, touching the iron to some things will grounded it and cause emi that causes resets in the pd controller.
> 
> I'm experiencing the same issue with the PinePower, with both my MHP30 hotplate and my TS100 using a Flex-C-Friend. Maybe a long shot, but is it possible to make the firmware tolerant to these kind of resets?

In my testing so far its the PD in the PinePower that is resetting, not the iron so this firmware has no control over it :(

gniezen, 2021-10-11:
> In my testing so far its the PD in the PinePower that is resetting, not the iron so this firmware has no control over it :(

I realize it's the PinePower that's at fault here, but I wast just wondering if the firmware could detect whether a reset has occurred and go back to the previous power setting instead of standby mode?

rblakeslee, 2021-10-16:
I'm encountering this same bug with a TS100 using several different adapters.
I don't think it's a power issue, I think it's thinking the movement is timing out or a button is being pressed to cancel.
Basically, power it on, set it to soldering mode, it heats up, but the very second it hits set temp, it starts flashing into cooldown mode (I have flash on cooldown set)
I can tell it to start again, but again, at set temp, it starts cooldown immediately. 
Auto-start behaves the same way, since it's not like a power loss or reset, or if it is, it's completely transparent because the screen doesn't flash off, or anything.

Ralim, 2021-10-16:
@rblakeslee 

If its an UVLO you should be presented with the warning text.
Can you confirm which firmware build you are running when this is occurring? Also what rating of power adapter are you using?

The intent is that the only way to exit soldering mode without seeing some form of warning is by the two button press or holding the rear button to exit. Other wise you should see some visual clue (uvlo warning / device goes to sleep first)

If you have a spurious reset from bad power supply you most definitely can have a reset that is hard to see. You can check this using the uptime counter in the debug menu (easiest is power up unit for a bit, then heat up, then when jumped back to menu check this time and see if its less than the expected value)

discip, 2022-04-23:
@BytheWWW
@howie-s
@2m
@gniezen
@rblakeslee
Is your issue still present, after updating to the latest release?
If not, please close this issue. 😊

thanks

discip, 2022-05-05:
Interesting my **Pinecil** works as intended with the latest firmware. 🤷‍♂️

Is your device *(**Pinecil**)* a genuine one?

Tried these:
Power Supply
|-
**[Anker Nano II 30W](https://de.anker.com/collections/wandladegerate/products/a2146?variant=42298016694435)**
**[Anker Nano II 45W](https://de.anker.com/collections/wandladegerate/products/a2664?variant=41566778851491)**
**[UGREEN 65W GaN Charger](https://www.ugreen.com/products/65w-gan-usb-c-fast-charger?variant=32127034654782)**

2m, 2022-05-06:
Yes, got it from the Pine Store. Previously in this thread we discovered that it is power supply related. And I am using my Pinecil with PinePower - 120W Desktop Power Supply - EU version, which seems to be the cause of these restarts.