Ralim/IronOS #774: Start in standby mode

HanaO00, 2021-01-06:
Hello, I wanted to try both options:
- SettingStartSleep
- SettingStartSleepOff

With the first, the device is supposed to start in standby mode and therefore go to the temperature designated in the options I think, but with the second, the temperature is supposed to go to room temperature, I put the TS100 down and set this option, but by turning on the iron, without moving it so that it remains in standby mode, the temperature rises to the same as the normal set temperature, is this normal?

discip, 2021-01-13:
@HanaO00
Good afternoon,
if you want to get help quicker, it would be nice, if you would let people know,
1. which **device** you are referring to (*I am aware of you mentioning the TS100. Just for integrity.*)
2. what **firmware** you are using,
3. which **PCB version** is present in your device &
4. what **power supply** you feed your iron from.

That way, the range of possible roots to the issue could be narrowed down.

<details>
  <summary>
    <b>example</b>
  </summary>
<h2><b>Copy the text below and change the values to those corresponding to your setup,</b></h2>

```c#
iron  |  Firmware Version  |  PCB Version  |  Power Supply
-|-|-|-
**TS100**  |  **https://github.com/Ralim/IronOS/commit/2052f308b09f7357d91e7aba80ffbda490fc6769**  |  **1**  |  **Powerbank_XY**
```

<h2><b>to make it look like this:</b></h2>

iron  |  Firmware Version  |  PCB Version  |  Power Supply
-|-|-|-
**TS100**  |  **https://github.com/Ralim/IronOS/commit/2052f308b09f7357d91e7aba80ffbda490fc6769**  |  **1**  |  **Powerbank_XY**
</details>

HanaO00, 2021-01-16:
iron  |  Firmware Version  |  PCB Version  |  Power Supply
-|-|-|-
**TS100**  |  **https://github.com/Ralim/IronOS/commit/8acfc7e587184d947c054d7e039961b10a3c3e0e**  |  **2**  |  **DC5525 - 24V**

discip, 2021-01-16:
@Ralim
After you managed to successfully tackle this issue for the **TS80P**, now the **TS100** needs to get a similar treatment as well. 😄

Ralim, 2021-01-16:
PCB version 2 has the same accelerometer and same code as your TS80P.

@HanaO00  Could you try the latest `master` build to confirm if the issue still applies?

HanaO00, 2021-01-17:
It works, thank you @Ralim & @discip

closed 2021-01-17 by Ralim