Ralim/IronOS #777: Not a Bug (don't think) more a question for the TS100 :)

Doyle4, 2021-01-07:
Bought a tip from Amazon, reviews where good.
But its the only tip that cant hold its temp, it fluctuates +5c, so 310 = 310c to 315c constant.

PID tuned, set DC for my 24V PSU

Not looking for a fix, just some tips on how to maybe control it a bit better or if this is a common issue with 3rd party tips? All new to me the TS100.

Doyle4, 2021-01-09:
Anyone have any information or tips on this please?

Vinigas, 2021-01-09:
**Do not bump the thread.**

Read all documentation first and try calibrating it and see if it fixes issue.
https://github.com/Ralim/IronOS/blob/master/Documentation/

Doyle4, 2021-01-09:
Done all them which is why im little lost with what do, I have also limited wattage with no change.

Ralim, 2021-01-09:
At 24V you may see fluctuations like that depending on how well the tips internal construction is.

This is not a cause for alarm, and doesn't affect soldering.

As this firmware takes the approach of never lying to users (unlike irons that filter this out), it does mean that you see everything as it occurs.

Doyle4, 2021-01-09:
Thanks for replying @Ralim Your answer has made me feel much better about this. Thanks!

closed 2021-01-09 by Doyle4