Ralim/IronOS #781: Tip electrical contacts

Svitsit, 2021-01-08:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug/Feature Request/Question/Complaint]_

* **I have**
  - [ ] Searched previous issues 
  - [ ] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [ ] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

* **What is the expected behavior?**

***Steps to reproduce the bug:***

- Turn on...
- Hold....

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 2.x?
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

The contacts for the sting were broken, I couldn't find where to buy them on the Internet, can you help? I apologize if I wrote in the wrong place, my English is not very well mastered, so I could write something wrong.

Ralim, 2021-01-08:
If you are referring to the metal contracts that connect to the tip when you put it in the handle.
You will need to contact the manufacturer of your iron, I don't believe any are available to buy online.

If English isn't your strong point you can try your preferred language as there are some multilingual people who reads these too.

Svitsit, 2021-01-08:
> Если вы имеете в виду металлические контакты, которые соединяются с наконечником, когда вы вставляете его в ручку.
> Вам нужно будет связаться с производителем вашего утюга, я не думаю, что его можно купить в Интернете.
> 
> Если английский не является вашей сильной стороной, вы можете попробовать свой предпочтительный язык, так как некоторые многоязычные люди тоже его читают.

Svitsit, 2021-01-08:
Ok, I will write to the manufacturer, thank you, I still hope that someone here will throw off the link to contacts, I will write in English with the help of a translator, I think it will be easier to work with people.

Svitsit, 2021-01-08:
If I find out something, I will leave an answer or a link here.

Doyle4, 2021-01-16:
I have a friend who does laser cutting, Im planning on sending him some to hopefully be able to reproduce some.
We are looking at Copper ones and 99.9% Pure silver.

Yes the silver is over the top, but some people like to have the best of the best :)

Ralim, 2021-01-16:
Ah be careful, you can't use silver for the contacts as the contacts touching the rings on the tip are part of the thermocouple setup. So changing the material properties will change the thermocouple circuit response and your temperature control will be out of whack :'(

Doyle4, 2021-01-16:
Thanks very much for that info! wasn't aware of that.

discip, 2021-01-17:
@Svitsit
Do you have any update on that yet?

discip, 2022-04-23:
@Svitsit
@Ralim
Since this is hardware related, this should be closed. 😊

thanks

Ralim, 2022-04-23:
ohh I forgot about this. For ts100/Pinecil Pine64 sells them now.
https://pine64.com/product/pinecil-copper-clips/

closed 2022-04-23 by Ralim