Ralim/IronOS #782: [Request] Enable Wiki Edit

jLynx, 2021-01-12:
Hi Ralim

Thanks for the awesome amount of work your have put into IronOS, its just amazing!

I am wanting to know if you can unlock the wiki as I would like to do some work on it if possible?

Thanks!

Ralim, 2021-01-12:
Hi,
Wiki is locked as people kept putting promotional links in for everything :)
I've been trying to move all of the documentation into the repo so that its git tracked as well.
Always open to PR's for documentation at any time without issue.

jLynx, 2021-01-12:
Ah I was more going to clean up the Wiki and formatting of everything as its a bit all over the place at the moment. Would you be able to make it editable for the next 30 minutes?

Ralim, 2021-01-12:
Go for it

jLynx, 2021-01-12:
You can lock it for now. I have done a bit of work on it. Still more that needs to be done which I can do later on :)

Ralim, 2021-01-12:
❤️ 
Happy to add people who are doing good work to the list of people allowed

jLynx, 2021-01-12:
Thanks very much. Will close this unless anyone else would like to help :)

closed 2021-01-12 by jLynx

Ralim, 2021-01-12:
Dude, your doing the hard work here looking after a wiki; dont need to thank me 🥇