Ralim/IronOS #785: TS100: if bricked or unable to change firmware <= solution!

sf88, 2021-01-15:
_This is a [Complaint]_

* **I have**
  - [x] Searched previous issues 
  - [ ] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
Cant change firmware
or
bricked after firmware change.

***Steps to solve it***

If bricked:
- download latest factory firmware, [MINI218.hex](http://www.minidso.com/forum.php?mod=attachment&aid=NjYzN3xmYWM0OTE4N3wxNjEwNjY5NjA0fDB8ODky)
- replace all letters in the name to capital: MINI218.HEX
- copy this to the TS100 and hopefully it will be OK again

If unable to change firmware:
- download the required firmware, like TS100_EN.hex
- replace all letters in the name to capital: TS100_EN.HEX
- REMOVE UNDERSCORE: TS100EN.HEX
- copy this to the TS100 and hopefully it will work

  - Firmware Version: 2.13
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) : 19.1/3

Vinigas, 2021-01-16:
Could you also specify official forum post, which has this link?

FrenzyIncarnate, 2021-03-17:
with DFU 3.45 I couldnt flash the latest firmware hex because the file extension was hex instead of HEX. this fixed the problem.. thanks

Psyc0Flyer, 2021-07-31:
I have DFU: 3.45 and I tried to also do the update and it just keeps rebooting. I have tried this method also and the same outcome. When I copy file to the USB folder of the TS100, it just reboots and the transfer says "An unexpected error is keeping you from copying the file. If you continue to receive this error, you can use the error code to search for help with this problem. Error 0x800701B1: A device which does not exist was specified."

There are 3 files on the TS100 when I open the folder.
.fseventsd
.Trashes
._.Trashes

Ralim, 2021-10-18:
Almost always its the computer not playing with the Miniware bootloader. Best bet is to use a different computer or operating system.

Max313233, 2021-12-15:
Hi, here is my problem:

sainsmart toolpac pro32

when connected to PC (tried several) the screen only showed DFU while I was holding Button A, drives were appearing and disapearing on the PC.

when Button released, screen says "settings", drive has a settings file, drive volume is limited to 8k
stupid me erased the settings file.

after this I was able to have the screen showing DFU constant but several win PCs didnt have a drive.

Now it is disasambled and connected to a STLINK
I can erase, flash, change optionbyte ...

screen does nothing, Win is showing  "unknown USB device" in device manager

I tried flashing several bootloader and fw files

is there a special set of option-bytes I have to setup?
which bootloaders could I test now?

thx 4 help

Ralim, 2021-12-15:
All you need to do is have option bytes as per factory defaults and flash this file https://github.com/Ralim/IronOS/blob/master/Development Resources/DO NOT FLASH_Flash Backup.hex

That is a working dump of a TS100's flash.

Max313233, 2021-12-15:
sry this is a 404 page
 https://github.com/Ralim/IronOS/blob/master/Development

Vinigas, 2021-12-15:
https://github.com/Ralim/IronOS/blob/master/Development%20Resources/DO%20NOT%20FLASH_Flash%20Backup.hex

paulfertser, 2021-12-15:
With SWD connected I would suggest to consider "Installing dapboot
from eDesignOSS (make TARGET=TS100 -C src) is recommended as it allows
reliable flashing of binary files with dfu-util."

Max313233, 2021-12-15:
it is all working now !
thx
I flashed the last stable Ralim with the bootloader, very nice !

discip, 2022-04-23:
@Ralim
Since the OP never answered and the issue is a known one, this one can also be closed.

closed 2022-04-23 by Ralim