Ralim/IronOS #791: Help Wanted | CJK Fonts

Ralim, 2021-01-16:
This has been talked about a lot, and the final state things got to was that adding support for Chinese/Japanese/Korean (CJK)  was that it wasnt possible to support the smaller fonts.

Now that we have the dynamic font tables, I believe it should be possible to have a solution of allowing CJK in large font, and just having to fallback in the smaller fonts to non-CJK.

I've started some cleanup to bring back the ability to use larger fonts for the menu's

Main blocking issue is having a fonttable for the large font (16 pixels high, 12 pixels wide).

I would love to open it to the community for any help with the following:

1. [ ] What font cant be used at this low resolution
2. [ ] Render this font out to 16x12
3. [ ] Once those are done ^ Testing how font looks on screen
4. [ ] Once that is good ^ Testing translations

alvinhochun, 2021-01-17:
CJK interface support will be exciting. I'll make sure to add a Cantonese (HK) interface, and a Chinese (Traditional, Taiwan) one if nobody else does it.

On the choice of font: As a starting point any CJK font is fine, which at least allow us to make things functional before finally fine-tuning the looks. In the end I expect each of the CJK languages to use different fonts due to regional variations. I will even consider manually adjusting the glyphs.

Is the current translation generation process smart enough to only include glyphs that are actually used and remap the character codes?

Ralim, 2021-01-17:
> 
> 
> CJK interface support will be exciting. I'll make sure to add a Cantonese (HK) interface, and a Chinese (Traditional, Taiwan) one if nobody else does it.
That would be fantastic ❤️ 
> On the choice of font: As a starting point any CJK font is fine, which at least allow us to make things functional before finally fine-tuning the looks. In the end I expect each of the CJK languages [...]
Do you know of any per chance that would be good to start with?
> 
> Is the current translation generation process smart enough to only include glyphs that are actually used and remap the character codes?
Yep, it generates a custom font table for every language build, so we can easily add as many glyphs to the font as we want and it will only generate those that are used. It creates a custom lookup for each language. At the moment this is limited to ~240 chars per language, but I'm _hopeful_ we can keep below that :) (If not will add extension fairly easily)

alvinhochun, 2021-01-17:
> Do you know of any per chance that would be good to start with?

I don't yet have a good suggestion. The only small size bitmap font that I find to be acceptable is MingLiU that came with Windows XP but obviously we can't use that.

For preliminary proof-of-concept testing, I would perhaps start with WenQuanYi Bitmap Song 9pt (downloadable at https://sourceforge.net/projects/wqy/files/wqy-bitmapfont/1.0.0-RC1/). However, it is stated to be licensed under GPLv2 while IronOS is GPLv3, so I'm not entirely sure whether it is fine to be embedded. (Then in the readme it states that it is licensed under "GNU Free Documentation License, Version 1.2 or any later version", so there is conflicting information...)

> At the moment this is limited to ~240 chars per language, but I'm hopeful we can keep below that :) (If not will add extension fairly easily)

I doubt that 8-bit will be enough. We will likely need 16-bit char codes or employ a DBCS or utf8-like encoding scheme. But let's just pretend we don't need that for now until we can't.

alvinhochun, 2021-01-17:
I think it'll be more complicated than just using a fixed size font table, so I'll try to experiment a bit to see what can be done.

alvinhochun, 2021-01-18:
So I gave it a try. While not ideal, it's actually not that bad, at least it can be serviceable for Chinese.

https://user-images.githubusercontent.com/2397650/104899715-26836f80-59b6-11eb-83c1-d0b8667b105b.mp4

A menu group can show 6 chars, but for some of the settings (those with numerical values) even the 5th char can get partially covered. This is ok-ish for Chinese, but when it comes to Japanese this might be a bit limiting. I'm going by the assumption that Japanese text tends to be slightly longer than Chinese due to the use of kana.

Adding a bit of auto text scrolling to the menu might work, though I'm not sure how complex it will be.

Also, it may be nicer to have a 16x16 font size (with Latin and numeric characters being 8x16) for the menu description. The large characters mixed in with the smaller CJK glyphs is a bit jarring.

Regarding the font licensing, I have sent an email to the font author to see what can be done.

I pushed my POC here: https://github.com/alvinhochun/IronOS/tree/cjk (You will need to manually download the font.)

Ralim, 2021-01-18:
I'm impressed at how readable that is.
Also amazed you figured out the mess of the fonts in ~24 hours

>This is ok-ish for Chinese, but when it comes to Japanese this might be a bit limiting.
I'm okay with starting down a track for a Chinese solution first; often when playing with these things getting a rough version is harder than tuning for a language later on. Having some groundwork is more important to me :) 

Main issue with having wider chars is the consumed flash space (some languages are already _very_ tight on the miniware units).
Though, could add an extra table that is wider for these fonts and maybe its only available on irons with enough room.. hmm..

If that font includes latin letters, could also swap out the entire font table when building for CJK for the large letters.
That would make all of the letters match each other?

Should be quite doable to extract the font dimensions into attributes that are produced from the python code, so that different languages could have different spacings.

>Adding a bit of auto text scrolling to the menu might work, though I'm not sure how complex it will be.
Certainly doable, its not _horrible code_ though my first thought would that, that would be jarring too?

alvinhochun, 2021-01-18:
> I'm impressed at how readable that is.

12px is readable for most commonly used glyphs, though it really is the smallest size that is practical.

> Also amazed you figured out the mess of the fonts in ~24 hours

The bitmap editor most certainly helped.

> I'm okay with starting down a track for a Chinese solution first; often when playing with these things getting a rough version is harder than tuning for a language later on. Having some groundwork [...]

I don't want to comment too much regarding Japanese or Korean because I don't *know* them well enough. It would be nice if some native Japanese or Korean speakers can chime in.

> Main issue with having wider chars is the consumed flash space (some languages are already _very_ tight on the miniware units).
> Though, could add an extra table that is wider for these fonts and maybe its only available on irons with enough room.. hmm..

I don't even know if the current font table will fit on the Miniware irons, given there will be almost double amount of glyphs or more than the current largest (UK with 120 glyphs).

> If that font includes latin letters, could also swap out the entire font table when building for CJK for the large letters.
> That would make all of the letters match each other?
>
> Should be quite [...]

The letters in WenQuanYi Bitmap Song are not fixed-width, so you will have to implement variable-width fonts for it to work. (Though if you do add variable-width fonts then I would consider using other Latin fonts and not the ones that comes with WenQuanYi Bitmap Song.)

Even for fixed-width CJK fonts, it's not exactly "fixed-width". Historically, CJK glyphs are square (full-width), while the ASCII characters are thin (half-width). This goes back to the early days of DOS, with the development of various double-byte character sets (DBCS) throughout the CJK computer industries. They for the most parts use two bytes for CJK characters, while maintaining compatibility with ASCII, so any ASCII characters only take one byte. Incidentally, double-byte CJK characters consumes the screen space of exactly two single-byte ASCII characters, simply due to how they were implemented.

This legacy still remains today:
[image]

So really the simplest way to get better text proportions is to make it so that CJK glyphs takes the space of two alphanumeric glyphs.

(But if you do want to support variable-width fonts, I won't stop you :P)

Frankly I think my POC is actually good enough for an initial version. I don't mind it too much.

> > Adding a bit of auto text scrolling to the menu might work, though I'm not sure how complex it will be.
> 
> Certainly doable, its not _horrible code_ though my first thought would that, that would be jarring too?

It will depend on the implementation. I have some ideas but I won't rush to implement them.

---

By the way, I completed the preliminary translation for Cantonese and it uses 242 glyphs, so I'm just going to guess that 8-bit char will be just barely enough for all three Chinese variants. So the only real blocking issue here should be the font licence. It's probably a non-issue and surely the authors won't mind it being used in a FOSS project, but I would just like to do it properly.

alvinhochun, 2021-01-20:
The [AR PL UMing](https://www.freedesktop.org/wiki/Software/CJKUnifonts/) font ([working mirror of release file](https://deb.debian.org/debian/pool/main/f/fonts-arphic-uming/fonts-arphic-uming_0.2.20080216.2.orig.tar.bz2)) could be a candidate. It contains 12px bitmap which I converted to BDF with FontForge and tested (missing several HKSCS glyphs but I drew out my own). However the [Arphic Public Licence](https://ftp.gnu.org/gnu/non-gnu/chinese-fonts-truetype/LICENSE) also seems incompatible.

alvinhochun, 2021-01-22:
The author of the WenQuanYi font has replied that he is fine with re-releasing the font using "GPLv2 or later versions", so it should be fine (but we still need to wait for the formal release).

@Ralim In the meantime, can you do a preliminary review on https://github.com/alvinhochun/IronOS/commit/e519ed2b1988acc2d88e49d6b155a0d959515eb9?

Ralim, 2021-01-24:
Having a look over the changes, these are my main thoughts; though all of these are up for discussion 😂

1. Could you use a git submodule for the external source rather than copying all of the files in?
1. I _feel_ it could be cleaner to make the OLED class aware of the start-with-newline-means-large-font
1. 1. As then we could move all of the font size changing into that class perhaps?
1. A later change (not at all required for a pr) would be to change the code that generates the large and small font tables to only require including symbols that are used in one or the other sizes. Though 100% happy for you to not worry about this and I can get to it :)

alvinhochun, 2021-01-24:
> Having a look over the changes, these are my main thoughts; though all of these are up for discussion 😂
> 
>     1. Could you use a git submodule for the external source rather than copying all of the files in?

Actually, only 4 out of 22 files in the release are copied over. There does not exist an authoritative git repo that hosts the release files so it is not an option anyway.

>     2. I _feel_ it could be cleaner to make the OLED class aware of the start-with-newline-means-large-font
> 
>     3. 1. As then we could move all of the font size changing into that class perhaps?

In my view, the caller of the `OLED` class has to set the font and then print, and it is for a good reason, as the caller knows best what font size it wants. Moving the responsibility of setting the font size to `OLED` seems counter-intuitive, no?

>     4. A later change (not at all required for a pr) would be to change the code that generates the large and small font tables to only require including symbols that are used in one or the other sizes. Though 100% happy for you to not worry about this and I can get to it :)

This is reasonable, and I think some ways of doing this may also help with making alphanumeric glyphs half-width (relative to CJK ones), but I also think this is something that can be done at a later time. Perhaps when someone wants to add Japanese or Korean translations, because they are inevitably going to exceed the current glyph count limit and this will require changes to font handling.

Ralim, 2021-01-26:
> Actually, only 4 out of 22 files in the release are copied over. There does not exist an authoritative git repo that hosts the release files so it is not an option anyway.

Ah gotcha, yeah thats fine then :)

> 
> > 1. I _feel_ it could be cleaner to make the OLED class aware of the start-with-newline-means-large-font
> > 2. 1. As then we could move all of the font size changing into that class perhaps?
> 
> In my view, the caller of the `OLED` class has to set the font and then print, and it is for a good reason, as the caller knows best what font size it wants. Moving the responsibility of setting the font size to `OLED` seems counter-intuitive, no?

Somewhat countr intuitive, but also wondering if adding the extra byte to the start of a string would be less than the jump into setting font each time; and could reduce code size. Not _entirely_ sure where things would fall on that though, and absolutely we can leave that for another day :)

> 
> > 1. A later change (not at all required for a pr) would be to change the code that generates the large and small font tables to only require including symbols that are used in one or the other sizes. Though 100% happy for you to not worry about this and I can get to it :)
> 
> This is reasonable, and I think some ways of doing this may also help with making alphanumeric glyphs half-width (relative to CJK ones), but I also think this is something that can be done at a later time. Perhaps when someone wants to add Japanese or Korean translations, because they are inevitably going to exceed the current glyph count limit and this will require changes to font handling.

Yeah, font handling has historically been a bit finnicky.

I'm happy to go with this for now :)

alvinhochun, 2021-02-03:
Update: The author of WenQuanYi Bitmap Song uploaded the font with the licence specified as "GPL v2 or later versions" so it should be good now. If I missed anything regarding the licence please let me know.

https://sourceforge.net/projects/wqy/files/wqy-bitmapfont/1.0.0-RC1/wqy-bitmapsong-bdf-1.0.0-RC1_GPLv2%2B.tar.gz/download

I will rebase the branch on top of master with the object file changes in a few days. But before that, I would like to have #824 handled so that I can include the license information of the font with it.

closed 2021-02-06 by Ralim

OctopusET, 2025-04-01:
https://littlelimit.net/misaki.htm

PR #845 (2021-02-06): CJK support with zh-TW and yue-HK translations
* **Please check if the PR fulfills these requirements**
- The changes have been tested locally
- There are no breaking changes

* **What kind of change does this PR introduce?**
Feature

* **What is the new behavior (if this is a feature change)?**
Initial support for CJK translations with Chinese (Taiwan, Traditional) and Cantonese (Hong Kong) translations included.

* **Other information**:

Closes #791

Note: This PR includes a commit from #844