Ralim/IronOS #793: Build Question

Doyle4, 2021-01-16:
Hello, Is it possible to compile a firmware on Visual Studio Code? If so, is it possible to know how please? I have imported it in but cant find a way to build.

Thanks :)

Ralim, 2021-01-16:
Hello,

So the easiest way is to use the start-dev script to launch a Docker container will all the tools installed and then attach visual studio code to that.

You can then use the make file to compile the code (or the build script).

Doyle4, 2021-01-16:
Thanks for your reply, I was just updating my question as you replied.

Seems much easier via terminal then :) 
How would I start the process? 
So far I have tried: 
Dragging Makefile into terminal = Error
Dragging Build.sh into terminal = Error
All above with make -j8 lang=TS100=EN model on the end = Error.

Sorry, still new to building things my self

Ralim, 2021-01-16:
Ah, so your on mac.

So the "simplest" (and cleanest cleanup later) is to use the docker container to create a small Ubuntu environment that vscode can attach to + has the compilers setup.
Below are a list of steps, not 1000% on these as I dont own any Apple devices, but

1. Install docker for mac + docker compose (if its not included)
1. Install the vscode extension "Vs code remote - containers"
1. Clone the source code somewhere if you haven't
1. In terminal change to the folder of the code
1. Run `./start_dev.sh` in the terminal
1. Docker will take a while to build the image and then drops you into a shell
1. Minimise this window
1. Open VS Code
1. Click the green icon in the lower left
1. Select attach to running container
1. Select the only one in the list (assuming this is your first time)
1. Vscode should spin for a bit and then connect you into the container
1. Open the folder in VS Code `/build/source`
1. You can now edit all of the source code as you wish
1. To build, in the terminal at the bottom of the window, you want to `cd` to `/build/source/Workspace/TS100/`
1. And then execute the `./build.sh` to build _everything_ or you can use the `make` command to build different models / languages.

Now, if you dont want to edit anything and instead _just_ compile.
You can instead just do steps 1,3,4,5,6, and then in the terminal window run steps 15/16

Ralim, 2021-01-16:
The main reason to use the docker image is that you need specific compilers depending on the model of the microcontroller. You can install these via brew i believe, but the dockerfile is known to work :)

Doyle4, 2021-01-16:
Thank you!
Once back home will try this.

Thanks again!

Doyle4, 2021-01-18:
Sadly no luck, will be honest and say I have no clue what Im doing, I never used Terminal for anything, downloaded Docker and Compose and getting constant errors.

Doyle4, 2021-01-18:
Last login: Mon Jan 18 09:59:35 on ttys000
Doyles-MacBook-Pro-2:~ Doyle$ /Users/Doyle/Desktop/IronOS-master/start_dev.sh 
ERROR: 
        Can't find a suitable configuration file in this directory or any
        parent. Are you in the right directory?

        Supported filenames: docker-compose.yml, docker-compose.yaml

Doyle4, 2021-01-18:
Last login: Mon Jan 18 10:01:26 on ttys000
Doyles-MacBook-Pro-2:~ Doyle$ /Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile 
: command not foundp/IronOS-master/workspace/TS100/Makefile: line 1: 
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 2: ifndef: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 3: model:=TS10: command not found
: command not foundp/IronOS-master/workspace/TS100/Makefile: line 4: endif
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 5: model: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 5: lang: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 6: ifndef: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 7: lang:=: command not found
: command not foundp/IronOS-master/workspace/TS100/Makefile: line 8: endif
: command not foundp/IronOS-master/workspace/TS100/Makefile: line 9: 
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 11: APP_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 12: MINIWARE_INC_CMSIS_DEVICE: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 13: MINIWARE_CMSIS_CORE_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 14: MINIWARE_HAL_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 15: MINIWARE_HAL_LEGACY_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 16: FRTOS_CMIS_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 17: FRTOS_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 18: MINIWARE_STARTUP_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 19: DRIVER_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 20: BSP_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 21: THREADS_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 22: MINIWARE_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 23: PINE_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 24: PINE_VENDOR_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 25: PINE_BOARD_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 26: PINE_VENDOR_USB_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 27: PINE_NMSIS_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 28: PINE_FREERTOS_PORT_INC_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 29: SOURCE_THREADS_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 30: SOURCE_CORE_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 31: SOURCE_DRIVERS_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 32: INC_PD_DRIVERS_DIR: command not found
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 33: SOURCE_MIDDLEWARES_DIR: command not found
: command not foundp/IronOS-master/workspace/TS100/Makefile: line 34: 
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 36: syntax error near unexpected token `$(model),$(filter $(model),TS100 TS80 TS80P)'
/Users/Doyle/Desktop/IronOS-master/workspace/TS100/Makefile: line 36: `ifeq ($(m'del),$(filter $(model),TS100 TS80 TS80P))
Doyles-MacBook-Pro-2:~ Doyle$

Ralim, 2021-01-18:
You cant just drag the file onto terminal
You need to use the `cd` command to change into the right folder first, then run `start_dev.sh`

Doyle4, 2021-01-18:
Thanks again for replying, and sorry for been a pain.

Last login: Mon Jan 18 10:09:05 on ttys000
Doyles-MacBook-Pro-2:~ Doyle$ cd /build/source/Workspace/TS100/
-bash: cd: /build/source/Workspace/TS100/: No such file or directory
Doyles-MacBook-Pro-2:~ Doyle$ cd /Users/Doyle/Desktop/IronOS-master/workspace 
Doyles-MacBook-Pro-2:workspace Doyle$ /Users/Doyle/Desktop/IronOS-master/start_dev.sh 
Traceback (most recent call last):
  File "site-packages/urllib3/connectionpool.py", line 677, in urlopen
  File "site-packages/urllib3/connectionpool.py", line 392, in _make_request
  File "http/client.py", line 1252, in request
  File "http/client.py", line 1298, in _send_request
  File "http/client.py", line 1247, in endheaders
  File "http/client.py", line 1026, in _send_output
  File "http/client.py", line 966, in send
  File "site-packages/docker/transport/unixconn.py", line 43, in connect
ConnectionRefusedError: [Errno 61] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "site-packages/requests/adapters.py", line 449, in send
  File "site-packages/urllib3/connectionpool.py", line 727, in urlopen
  File "site-packages/urllib3/util/retry.py", line 403, in increment
  File "site-packages/urllib3/packages/six.py", line 734, in reraise
  File "site-packages/urllib3/connectionpool.py", line 677, in urlopen
  File "site-packages/urllib3/connectionpool.py", line 392, in _make_request
  File "http/client.py", line 1252, in request
  File "http/client.py", line 1298, in _send_request
  File "http/client.py", line 1247, in endheaders
  File "http/client.py", line 1026, in _send_output
  File "http/client.py", line 966, in send
  File "site-packages/docker/transport/unixconn.py", line 43, in connect
urllib3.exceptions.ProtocolError: ('Connection aborted.', ConnectionRefusedError(61, 'Connection refused'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "site-packages/docker/api/client.py", line 205, in _retrieve_server_version
  File "site-packages/docker/api/daemon.py", line 181, in version
  File "site-packages/docker/utils/decorators.py", line 46, in inner
  File "site-packages/docker/api/client.py", line 228, in _get
  File "site-packages/requests/sessions.py", line 543, in get
  File "site-packages/requests/sessions.py", line 530, in request
  File "site-packages/requests/sessions.py", line 643, in send
  File "site-packages/requests/adapters.py", line 498, in send
requests.exceptions.ConnectionError: ('Connection aborted.', ConnectionRefusedError(61, 'Connection refused'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "docker-compose", line 3, in <module>
  File "compose/cli/main.py", line 67, in main
  File "compose/cli/main.py", line 123, in perform_command
  File "compose/cli/command.py", line 69, in project_from_options
  File "compose/cli/command.py", line 132, in get_project
  File "compose/cli/docker_client.py", line 43, in get_client
  File "compose/cli/docker_client.py", line 170, in docker_client
  File "site-packages/docker/api/client.py", line 188, in __init__
  File "site-packages/docker/api/client.py", line 213, in _retrieve_server_version
docker.errors.DockerException: Error while fetching server API version: ('Connection aborted.', ConnectionRefusedError(61, 'Connection refused'))
[50994] Failed to execute script docker-compose
Doyles-MacBook-Pro-2:workspace Doyle$

Doyle4, 2021-01-18:
I think it's best I just download compiled versions.

Vinigas, 2021-01-18:
Your Docker deamon is not running. Start Docker deamon first.

Doyle4, 2021-01-18:
> Your Docker deamon is not running. Start Docker daemon first.

Thank you, will give this a try.

closed 2021-05-08 by Ralim