Ralim/IronOS #794: Investigate/document compilation with standard compilers

t-8ch, 2021-01-16:
(Follow up from discussion in https://github.com/Ralim/IronOS/pull/792 )
(This ticket is written from the Pinecil perspective, but the same holds true for other targets)

Currently IronOS for Pinecil is built via a gcc provided by nuclei.

Switching over to a vanilla upstream compiler would provide various benefits:
* Users don't have to download a binary-only compiler from a third-party.
* IronOS can be sure to have receive new compiler/newlib features and fixes directly from upstream.

At least ArchLinux has compiler/newlib packages that can compile IronOS for Pinecil:
* Required packages: `riscv64-elf-gcc`, `riscv64-elf-newlib`
* Build invocation: `make model=Pinecil COMPILER_PREFIX=riscv64-elf`

Comparing the sizes of `Pinecil_EN.bin` from the v2.13 GitHub release and custom built code:
48432 Bytes for the official release (assuming to be built with the nuclei compiler)
48468 Bytes for "riscv64-elf-gcc (Arch Linux Repositories) 10.2.0"

Ralim, 2021-01-24:
I think to "move forward" on this, main things would be:

1. Ubuntu 20.04 shipping up to date compilers (untested) as thats whats used on the Github CI
2. Regression testing to make sure nothing breaks

I would prefer to use the upstream compiler if we can.

Also of note, the Nuclei compiler _is_ [open source](https://github.com/riscv-mcu/riscv-gnu-toolchain), just hard to find :)

t-8ch, 2021-01-24:
About 1)
Ubuntu 20.04 does ship version of a suitable gcc: https://packages.ubuntu.com/focal/gcc-riscv64-unknown-elf
However they are not shipping a matching binary version of newlib with it.
I opened a [bug with Debian](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=980918) to provide such a package.
After it is in Debian maybe it can be backported to Ubuntu 20.04. (At least it will be in future versions of Ubuntu)

In theory it should also be possible to vendor newlib, I don't think it is worth it.

Ralim, 2021-01-26:
Aha,

I agree that vendoring newlib would not be ideal.

Would like to see what happens with debian first; though pulling in a precompiled newlib and using github cache for it could work for CI wise.

Ralim, 2022-02-15:
I ended up solving this using the `alpine` base image as they package enough for this to all work :)

closed 2022-02-15 by Ralim