Ralim/IronOS #796: Custom boot logo appears mangled/distorted on TS80P

wlhlm, 2021-01-16:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

_This is a [Bug]_

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

Custom boot logos appear mangled using GH Actions [build](https://github.com/Ralim/IronOS/actions/runs/489962837) of (latest) commit https://github.com/Ralim/IronOS/commit/6d56ba3322959afdac235f8c955cfc69564f2a9c.

* **What is the expected behavior?**

The logo should display correctly. Flashing back release version 2.13 (without reflashing the logo) shows the logo correctly on boot.

***Steps to reproduce the bug:***

1. generate logo from PNG using the included `img2ts100.py` script
2. upload logo as hex file
3. reboot soldering iron
4. *boot logo shows up distorted*
5. flash release build (version 2.13)
6. reoot soldering iron
7. *boot logo shows up correctly*

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: 6d56ba3322959afdac235f8c955cfc69564f2a9c (broken), 2.13 (working)
  - PCB Version: unknown
  - Power Supply (Voltage and Current Rating): 5V normal USB power from host computer

[logo.zip](https://github.com/Ralim/IronOS/files/5824404/logo.zip) (includes source logo PNG and generate HEX files)

Ralim, 2021-01-16:
Ah heck.

I'll jump on this tomorrow

Ralim, 2021-01-16:
Should be fixed in new `master`.
my bad 😰

wlhlm, 2021-01-17:
Can confirm working. Thanks for the quick fix!

closed 2021-01-17 by wlhlm