Ralim/IronOS #797: Sorry another question, ksger t12

Doyle4, 2021-01-16:
Can this OS be installed on a ksger t12 (2.1S)

I know it has great features already, but iv found they are releasing new boards instead of firmwares now.
Would love Ralim/iRonOS on it.

HanaO00, 2021-01-16:
Hi, the firmware cannot be installed on the T12, only : 
* TS100
* TS80/TS80P
* Pinecil

As far as I know, there is no discord (yet)

Doyle4, 2021-01-16:
Thanks for the reply @HanaO00 

Cool, I wasn't too sure as couldn't find anything online thanks.

I hope there is one eventually, may help in the long run, in a couple 3D Printer ones and had great help/advice as there other advance users, people who have had problems etc and fixed.

I don't mind creating one if need be, like a 3rd Party version, could take some load from people maybe?

closed 2021-01-16 by Doyle4

Ralim, 2021-01-16:
At the moment this repo is focused on handheld units due to its focus on the small LCD layout and dual buttons.

It could be possible to support but that would take time I do not currently have.

Doyle4, 2021-01-17:
All good, appreciate you getting back to me.