Ralim/IronOS #798: Custom boot logo - simpler conversion to binary format

OndroNR, 2021-01-16:
_This is a [Question/Proposal]_

What is purpose of repeating boot logo bitmap until INTELHEX_MINIMUM_SIZE is reached? I would like to use lightweight `hex2bin.py` instead of tool from RISCV toolchain, but `hex2bin.py` complains about data overlap. Removing duplicated data and converting with `hex2bin.py` results in same binary file as using `riscv64-unknown-elf-objcopy`. Even better would be if `img2ts100.py` produced binary file directly.

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

Boot logo bitmap is repeated 4 times in .hex file. `hex2bin.py` tool complains about data overlap.
`ERROR: bad HEX file: Hex file has data overlap at address 0x800F800 on line 66`

* **What is the expected behavior?**

Boot logo bitmap is present only once in .hex file

* **Steps to reproduce the bug:***

1. `python3 img2ts100.py 001_PINECIL.png 001_PINECIL.hex`
1. `hex2bin.py 001_PINECIL.hex 001_PINECIL.hex.hex2bin`

* **What is the motivation / use case for changing the behavior?**

Using `hex2bin.py` from `intelhex` python package instead of tool from full RISCV toolchain.

* **What are you running:**

macOS Mojave, Pinecil

Ralim, 2021-01-16:
Hello,

The duplication is because of the miniware devices, as their bootloader has a minimum file size check.

It should be quite doable to just add a flag to the python code to directly output the binary however.

Ralim, 2021-01-16:
Addionally, you can use basically any objcopy to do this as far as i'm aware so you could use any others you have on your system already for this. Using the riscv one is just reccomended as you most likely already have it if you are messing with the Pinecil firmware.

OndroNR, 2021-01-17:
Well, for custom boot logo I don't need to compile firmware. I will look into modifying `img2ts100.py`, so it could output binary directly. Should it be format flag, or autodetection by file extension?

OndroNR, 2021-01-19:
Just more thoughts about Intel HEX and Miniware:
- does repeating boot logo binary in hex file wear out flash memory area faster?
- could we replace duplicate content with spammy "no-op" records, like "Extended Linear Address" or "End Of File" (more than once at the end, against the spec)

Ralim, 2021-01-19:
>does repeating boot logo binary in hex file wear out flash memory area faster?
Unknown; have not decompiled their bootloader.
I would expect that yes it does a few more cycles out of the thousands we have. People tend to reflash the firmware more than the logo.

>could we replace duplicate content with spammy "no-op" records, like "Extended Linear Address" or "End Of File" (more than once at the end, against the spec)?

You are welcome to try, but given how amazingly bad their bootloader is, I would suspect that we cant.

cybernesto, 2022-01-14:
I think this issue got solved with #1178. Can it be closed?

closed 2022-02-04 by Ralim