Ralim/IronOS #802: Custom Logo isnt shown

Pichi345, 2021-01-17:
This is a Bug

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**
If I upload a HEXwirh an boot pcture, it don't shows it and in soldering modes there are various pixels on the screen. If I reupload the Firmware, the pixels disapear but there is no startup logo. I'm working on a raspbian machine. It always gets .EER the first time but the second thime it alwaysworks.

* **What is the expected behavior?**
Show startup screen

***Steps to reproduce the bug:***
Install Ralim Firmware on a ts100 and upload a custom startup screen to it.

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

  - Firmware Version: 2.11.372F8E3
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) : 5s Lipo batery

Ralim, 2021-01-17:
You are most likely using a logo that was generated a long time ago and has a different storage offset than is used in the newer firmware releases.

Regenerating the logo with the newer python script will resolve this.

Pichi345, 2021-01-17:
Ok, I will try it. Thanks a lot!!!

closed 2021-01-24 by Ralim