Ralim/IronOS #803: Pinecil accelerometer not detected on v2.14

OndroNR, 2021-01-17:
_This is a [Bug]_

Works with v2.13 which I flashed after to test hardware.

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

No accelerometer detected message. Sleep countdown missing.

* **What is the expected behavior?**

Sleep countdown present, no warn message.

***Steps to reproduce the bug:***

- Flash v2.14 firmware
- Turn on
- See "No accelerometer detected" message
- Start heating, notice missing countdown

* **What is the motivation / use case for changing the behavior?**

-

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware & PCB version.
If using a USB-C supply please note model number

  - Firmware Version: v2.14
  - PCB Version: first production batch
  - Power Supply (Voltage and Current Rating) : USB QC 9V 1.67A

Ralim, 2021-01-17:
Hello,

This is not good to hear, the firmware worked on my sample unit from the first production batch.

Can you please confirm the PCB version that is shown in the debug menu on the new firmware?

Ralim, 2021-01-17:
Ah looks like a race, working on it now

Ralim, 2021-01-17:
Can you please test this once it finishes building: https://github.com/Ralim/IronOS/actions/runs/491373542

OndroNR, 2021-01-17:
Yes, I will

pfeerick, 2021-01-17:
I'd just grabbed mine to test this, and the updated binary was already up, so tried that instead since the 2.14 binaries have been pulled. I'm not getting any accelerometer errors on two batch 1 units.

Ralim, 2021-01-17:
Fantastic.
I think it was just the poll for the accelerometer coming too fast after power on.

closed 2021-01-17 by Ralim

OndroNR, 2021-01-17:
Works fine, thanks.

OndroNR, 2021-01-17:
Also PCB 3 in the debug menu