Ralim/IronOS #805: Pinecil freezes with custom boot logo on v2.14

OndroNR, 2021-01-17:
_This is a [Bug]_

Custom boot logo appears, but Pinecil does not react to buttons. Same boot logo binary works on v2.13.

* **I have**
  - [x] Searched previous issues 
  - [x] This is in _this_ firmware, not vendor "offical" firmware
  - [x] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

Pinecil freezes with custom boot logo shown.

* ***Steps to reproduce the bug:***

- Flash v2.14
- Flash custom boot logo
- Turn on
- See boot logo
- Try press buttons - no reaction

* **What are you running:**

  - Firmware Version: 2.14
  - PCB Version: First production batch
  - Power Supply (Voltage and Current Rating) : USB QC 9V

Ralim, 2021-01-17:
Hello,

Could you confirm if it times out or if the unit has frozen?

OndroNR, 2021-01-17:
I think it has frozen, but I am not 100% sure. Anyway, it works after fixing #803.

discip, 2021-01-17:
@OndroNR
If fixed, please close.
thanks

Ralim, 2021-01-17:
Ill close this all good.
I believe what was happening was actually the #803 bug was causing the accelerometer to lockup the bus.

closed 2021-01-17 by Ralim