Ralim/IronOS #810: Adding Thermocouples to Tips

dhiltonp, 2021-01-18:
I've built 3 tips with thermocouples hard-soldered/brazed to the end (670C melting point): https://youtu.be/oAAN9YCJlkc

I've got a miniware D24, a generic B2 tip and a Hakko T15-D2.

Here's my "logging" the thermal performance of the B2 under 2.14.1 and 2.12b. I knew 2.14.1 was better thermally, but I didn't realize by how much!

2.12b: https://www.youtube.com/watch?v=03S4dfgHxQY
2.14.1: https://www.youtube.com/watch?v=HL8qnq7Hi8Y

I haven't done any dynamic tests yet. I have run all 3 tips with both firmwares. 

The Hakko tip seems to be very accurate and responsive with both firmwares. This may be because it's a fat tip? 

2.14.1 is much better for the B2 and D24.  The achieved temperature much closer to the set temperature - I'm reading 5-10C over for the D24 and 10-15C under for the B2?. With 2.12b the D24 is ~30C under temp and the B2 is ~40C under temp!

------------

My logging hardware isn't complete - I've got to put in isolation because ground loops are an issue. So far I've just unplugged the laptop, but that won't work when I'm also collecting logs from the Pinecil!

dhiltonp, 2021-01-18:
I re-tested with 2.14.1 with the 3 tips. The B2 is 5C over, the D24 and D2 are reading about 20C over. Thermal lag on the D24 and D2 is about 2s, the B2 is about 5 seconds.

I did touch the D24 to some solder and found it worked well - the temp quickly dropped to the set point, and when touching the iron to a thermal mass wattage/temp compensation is definitely present but can be improved - the tip temp was reading 20-30C low.

Roughly speaking, our temp correction code is linear with wattage but should be exponential.

dhiltonp, 2021-01-18:
Hmmm - there is no way that the wattage correction code alone is responsible for that much drop, I'd expect to see a maybe 2-5*C correction at idle. There have been too many updates to the temp code to pin the changes on any one factor.

dhiltonp, 2021-01-18:
21e3bab8d9e3b9ead3c3a990a2594d49049f2948 or a3f037fd1d1de4a77200a55cb59df53105234db1 or   27bf2a17110373cb1a30bed259ac2fbc1b374c5c or 627c4919989ad447f96f451d7d7d3e0c9e0ac4c0?

Firebie, 2021-01-19:
`I re-tested with 2.14.1 with the 3 tips. The B2 is 5C over, the D24 and D2 are reading about 20C over. `
This can be related to different calibration offset 'systemSettings.CalibrationOffset' - for different original (miniware) tips I have tested - it varies from 650 to 950.

discip, 2022-04-23:
@dhiltonp
I think this *issue* fits better in the discussions section.
So maybe you could transfer it there and close it here? 😊