Ralim/IronOS #811: TS100 spinning DC barrel jack, produces sound when shaken.

SuperNeko, 2021-01-18:
_This is a [Question]_

* **I have**
  - [x] Searched previous issues 
  - [ ] This is in _this_ firmware, not vendor "offical" firmware
  - [ ] This is not a bug in the vendor bootloader (aka DFU)
  - [x] I have checked this is not already covered in the docs in `/Documentation`

* **What is the current behavior?**

Inside DC barrel jack has some metal "sleeve" and after two months of using soldering iron, it suddenly started to spin and now iron produces sound when shaken. Also, now my cable can rotate independently from soldering iron, same goes for soldering iron. 

* **What is the expected behavior?**

Before nothing produced noise when shaken and nothing spin. 

  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 24V 3A
  
  
When I purchased TS100, nothing behaved like this, but suddenly this happened. So my question is - is it normal behavior? I found few references to similar issues on the Internet, yet nothing useful, seller says that it is normal and apparently not a defect. It is still working fine even when rotating while working, but I am worried about issue in long-term. 
<details>
<summary>Image of DC barrel jack</summary>

[image]
</details>

Ralim, 2021-01-18:
These kinds of DC jacks use that sleeve to apply pressure to the outside of the barrel connector.
They are made of a pressed metal and are normaly held in place but they can come loose like this.
If its staying inside the jack its probably "ok" but not ideal.
Will probably fail at some point down the track, but this is "normal" wear and tear on these kinds of plugs really.
They all have a finite life.

However if the unit is making a noise I would open the back cover to check whats loose, as a loose piece of metal can kill the unit easily if it gets into the wrong spot.

SuperNeko, 2021-01-18:
Well, this sleeve is the only thing that shakes (besides buttons), I checked this when trying to find culprit for rattle. Thanks for your answer, hope my jack won't die soon.

closed 2021-01-18 by SuperNeko

paulfertser, 2021-01-18:
On Mon, Jan 18, 2021 at 12:59:09AM -0800, SuperNeko wrote:
> Well, this sleeve is the only thing that shakes (besides buttons),

FWIW, my unit also produces that rattling sound, but only when the
jack is not inserted. I think it's just normal for this connector.

-- 
Be free, use free (http://www.gnu.org/philosophy/free-sw.html) software!
mailto:fercerpav@gmail.com