Ralim/IronOS #813: Double tap to sleep/wake?

DerSpatz, 2021-01-19:
Would it be possible to add a "double tap to sleep/wake" function?

When you tap the tip of the iron two or three times onto a surface, it goes into sleep mode instantly, and then either waits for the next tap event to wake up again, or after tapping it goes into sleep and then waits 10 seconds before any movement wakes it up again, so you can place it down after tapping.

Could be combined with a lower sleep temp, too...

Ralim, 2021-01-19:
Hmm,

Yes this probably could be implemented.

I'll have to have a look into this and the best way to implement this but could possibly use the accelerometer tap detection.

digitalis-ts100, 2021-03-15:
HI,

This is a nice idea. :) 
Maybe if we point the tip up. it will go sleep mode. 
Like for ex. i hang my ts100 if i not use it. i need to wait for the timeout before it sleeps.
just an idea. tnx